mavlink/MAVSDK #24: [core] fool proof installation package

mrpollo, 2017-08-21:


julianoes, 2018-07-23:
We already generate .deb and .rpm for Ubuntu and Fedora. However, we also need to have an easy story for macOS and Windows. And it all needs to be downloadable from the docs.

hamishwillee, 2018-07-24:
> We already generate .deb and .rpm for Ubuntu and Fedora. However, we also need to have an easy story for macOS and Windows. And it all needs to be downloadable from the docs.

1. Where are they uploaded
2. How are they named 
3. Is there a version for each release?
4. Is there a version for latest.
5. Do these just install the headers and binaries in the same places as the normal system-wide install after build?
   * Do users need to update the linker cache after installing? : `sudo ldconfig` 
6. Are these just the libraries or do the packages include example code?
7. Is there a plan for MacOS/Windows binaries?

hamishwillee, 2018-07-26:
Thank you.  Re:

> The packages install to /usr/lib / /usr/include. The default is build install is `/usr/local/lib/` `/usr/local/include`. That's common practice.

Great ... but what if you have installed package and built system package. How does system know which to use? Or to put it another way, this app cmake file is for the SDK [built and installed system wide](https://sdk.dronecode.org/en/guide/toolchain.html#sdk_installed_system_wide) - what would this look like if using SDK installed from app package?

julianoes, 2018-07-26:
It depends on where cmake looks for includes and what the paths for the linker are. This is a general problem with libraries, not something that we can fix with this one, I believe.

julianoes, 2018-07-27:
@hamishwillee right, so I generally don't use the binary downloads because I already have it built and installed manually.

So, presumably the "user" documentation should say to install the lib, and the "dev" documentation would also mention how to build and install yourself which is what we have for now.

hamishwillee, 2018-07-27:
Exactly, so to wrote that user documentation, I need to understand how you then link to it. So we circle back - how can we get our current examples to work with a binary download?

julianoes, 2018-07-27:
Right, ok so it should just work because the linker by default checks `/usr/lib/`. On some systems `/usr/local/lib` is not even checked by default unless it's added to the `LD_LIBRARY_PATH` env variable.

So, if I install it manually in Fedora for instance, I need to do:

```
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/lib
```

hamishwillee, 2018-08-01:
@julianoes I can't document this yet, because the [current release](https://github.com/Dronecode/DronecodeSDK/releases) (0.5) installs dronecore libraries rather than dronecode_sdk. 

When do you plan doing a new release as "dronecode SDK"?

So far the only dependency I've found for Ubuntu 16.04 "out of the box" is that you still need to install cmake to actually use the file `sudo apt install cmake -y`

julianoes, 2018-08-02:
Ok, I'll do a new release ASAP.

The dependency of cmake only comes from the fact how you build your example. The library is just a C++ library, and you can use whatever build system to build against it.

hamishwillee, 2018-08-03:
Thanks. If you haven't seen an accompanying docs release a couple of days later, let me know. I missed it!

julianoes, 2019-08-06:
Mostly getting there, closing.

closed 2019-08-06 by julianoes