mavlink/MAVSDK #32: Feature: Add support for vision estimate feed

TSC21, 2017-08-21:
This would be a good enhancement for more complex systems

julianoes, 2018-07-24:
@TSC21 can you please give some more details what exactly is needed here. What data and functionality needs to be exposed? What are the underlying mavlink messages?

TSC21, 2018-07-24:
Sure thing. So the idea would be to interface this with using fast-RTPS as the Pub/Sub framework. A direct OpenCV visual odometry plugin could be written and be taken as an example, but wouldn't actually serve as an interface, but rather as an idea of how OpenCV could be integrated in a visual odometry pipeline.
The underlying Mavlink msgs would be `VISION_POSITION_ESTIMATE` (for pose), `VISUAL_SPEED_ESTIMATE` (for velocity) and `ODOMETRY`, which gives a more complete visual odometry msg.

TSC21, 2018-07-24:
The interface will be between the SDK and fast-RTPS. It doesn't really have to be between fast-RTPS and the PX4 internals. The use case is to use it to send VIO data to the autopilot, parsed from a fast-RTPS publisher.

TSC21, 2018-07-24:
@julianoes it doesn't have to be mandatory to interface with fast-RTPS, since we could rather use other types of Pub-Sub/message broker frameworks to the same purpose. For example, for the Android case, one can use [`EventBus`](http://greenrobot.github.io/EventBus/). [RabbitMQ](https://www.rabbitmq.com/) has even a [client for Swift](https://github.com/rabbitmq/rabbitmq-objc-client/). It does depend on what message brokers we do want to support.

julianoes, 2018-07-24:
Ok got it. In this case the SDK does not need to implement any of `fast-RTPS/RabbitMQ/etc` but only provide the API for mavlink which makes sense.

TSC21, 2018-07-24:
> In this case the SDK does not need to implement any of `fast-RTPS/RabbitMQ/etc` but only provide the API for mavlink which makes sense.

Partially. I mean, if you want to subscribe to published VIO data in the `RabbitMQ` framework, you need to add an handler for that data, right?

TSC21, 2018-07-25:
Let me know if you are interested to tackle this together.

julianoes, 2020-11-20:
There is support for VISION_POSITION_ESTIMATE in the Mocap plugin.

closed 2020-11-20 by julianoes