mavlink/MAVSDK #33: Feature: add support for ADS-B data

TSC21, 2017-08-21:
Useful for the supporting complete avoidance systems

julianoes, 2018-07-24:
@TSC21 can you please give some more details what exactly is needed here. What data and functionality needs to be exposed? What are the underlying mavlink messages?

TSC21, 2018-07-24:
For this case:
1. If we are outputting the data from Mavlink `ADSB_VEHICLE`, then this would have to be plugged to a Pub/Sub framework, like fast-RTPS, to have the data available to any other app, unless we plug this directly to a specific example where we could should the transponder data (for example, an Android app that would track the position of vehicles in the surroundings) - but any example would only serve as that - an example - because it would be application specific. For the case of DronecodeSDK, we probably want it serving as a bridge for this kind of data.
2. If we are connecting a transponder to our system and want to output the data to the flight controller using `ADSB_VEHICLE` Mavlink msg, then I guess we could be hardware specific and plug this to a driver that interfaces with the transponder.

julianoes, 2018-07-24:
Ok, so the `ADSB_VEHICLE` mavlink message goes both ways? From a transpoder to the autopilot and from the autopilot to a transponder?

TSC21, 2018-07-24:
Yes that's the idea. Currently there's a uORB topic called `transponder_report` which propagates the data of a transponder connected to the autopilot directly. Though, as we already do with `distance_sensor`, we can plug the transponder on an onboard or external system and send the transponder data to the autopilot, adding the handler on the `mavlink_receiver`.

DoppleGangster, 2021-03-29:
Looks like it has been a few years since this topic have been discussed so I figured I might as well bring it back from the dead. I'm guessing this feature has no current plans for integration? I was looking at adding this feature to the project I am working on so that I can display the current flight positions of obstacles in the area.

TSC21, 2021-03-29:
@TSC21 I might eventually find some time to work on this, but if you have an hard dependency on this, you can always have a look at the MAVSDK docs on how to add a new plugin: https://mavsdk.mavlink.io/develop/en/contributing/plugins.html.

DoppleGangster, 2021-03-29:
I'll take a look. Would you expect this to be a whole new plugin or just add to an existing one such at telemetry data?

TSC21, 2021-03-29:
> I'll take a look. Would you expect this to be a whole new plugin or just add to an existing one such at telemetry data?

I would say a new plugin because this is specific to a transponder, not vehicle internal sensors / telemetry.

DoppleGangster, 2021-03-29:
👍

TSC21, 2021-04-07:
Addressed in https://github.com/mavlink/MAVSDK/pull/1390.

closed 2021-04-07 by TSC21