mavlink/MAVSDK #50: Why is Loiter Mode Set request sent before Arm and Takeoff commands ?

Rjasuja, 2017-09-06:
For Aero board, Loiter mode set request fails and therefore, arming fails.
What is the requirement to set Loiter mode (specifically)?

julianoes, 2017-09-06:
This was done to prevent the vehicle from doing anything stupid on arming. For instance if it is in mission mode and you arm, it might not just spin the propellers but takeoff straightaway. This can be a safety hazard and the goal was to make arming and takeoff to explicit commands.

Do you know the reason why loiter/hold mode fails? Could it be that you try to arm without GPS?

hamishwillee, 2017-09-06:
>  it might not just spin the propellers but takeoff straightaway.

It **will** take off straight away on a copter or VTOL in almost all circumstances: https://docs.px4.io/en/flight_modes/mission.html

julianoes, 2017-09-06:
Right, which is why I prefer to have it in HOLD where it won't do anything if landed.

sugnanprabhu, 2017-09-06:
@Rjasuja issue could be because the testing is being performed indoor and the GPS signals are not available.

julianoes, 2017-09-06:
We should probably specify that DroneCode only supports flying given a position solution such as GPS.

hamishwillee, 2017-09-06:
When you say "such as", are you excluding other positioning methods - optical or whatever? I imagine we don't want to exclude indoor flight via DroneCore. Also if someone is using DroneCore for the purpose of integrating a camera subsystem (or similar) we don't want to prevent them from doing so just because we don't have GPS.

Rjasuja, 2017-09-07:
there is no GPS signal here. For that reason, arming fails. 
Can we set it to some other mode so as to remove the dependency on GPS ?

julianoes, 2017-09-07:
@hamishwillee 

> When you say "such as", are you excluding other positioning methods - optical or whatever?

I meant "as for instance GPS". Vision, flow should definitely also be supported.

> Also if someone is using DroneCore for the purpose of integrating a camera subsystem (or similar) we don't want to prevent them from doing so just because we don't have GPS.

But you don't need to arm a camera subsystem, do you? 😄

julianoes, 2017-09-07:
@Rjasuja 

> there is no GPS signal here. For that reason, arming fails.

For testing, you can just fake GPS. Just get a NuttShell (`nsh>`) and do:

```
gps stop
gps start -f
```

julianoes, 2017-09-17:
Closing because I think the question has been answered.

PR #886 (2019-10-24): Remove switching to HOLD mode before arm and takeoff
This removes most of the workaround where we switch to HOLD before arming. This was mostly out of precaution but as discussed in #50 it can hopefully safely get dropped now.

This also includes some refactoring and cleanup of the action plugin.

For reviewing, please check the individual commits and messages.

Fixes #50, fixes #614, fixes #855.
closed 2017-09-17 by julianoes