mavlink/MAVSDK #54: Dockerfile on dronecore/dronecore is out of date

hamishwillee, 2017-09-07:
This does not have doxygen - it is about a week old.

Would also be good to have a link back to the repo in https://hub.docker.com/r/dronecore/dronecore/

julianoes, 2017-09-07:
I just checked this and found that doxygen is included and that I have pushed it:

```
docker push dronecore/dronecore
The push refers to a repository [docker.io/dronecore/dronecore]
43dc7cb93aa8: Layer already exists
29006b6830e2: Layer already exists
56827159aa8b: Layer already exists
440e02c3dcde: Layer already exists
29660d0e5bb2: Layer already exists
85782553e37a: Layer already exists
745f5be9952c: Layer already exists
latest: digest: sha256:a8352de2ade873eda8f94840e3c2e90b500d0b07f7a20c40c3a695808b9c9d19 size: 1777
```

I already pushed the updated one with doxygen when the comments were still a pull request. This is what might have confused you.

julianoes, 2017-09-07:
Backlinks created, closing.

closed 2017-09-07 by julianoes

hamishwillee, 2017-09-07:
Thank you. Don't know what happened, but looks great now.