mavlink/MAVSDK #55: A picture is captured always during a mission execution though its not requested by application

shakthi-prashanth-m, 2017-09-07:
I modified [fly_mission](https://github.com/dronecore/DroneCore/blob/master/example/fly_mission/fly_mission_dronecore.cpp) example to add two mission items with NO Camera action. But 1 image was captured during mission execution.
I see that in [`void MissionImpl::assemble_mavlink_messages()`](https://github.com/dronecore/DroneCore/blob/master/plugins/mission/mission_impl.cpp#L310), command [MAV_CMD_DO_MOUNT_CONTROL](http://mavlink.org/messages/common#MAV_CMD_DO_MOUNT_CONTROL) is added in `message_gimbal`. Could this be the reason for capturing atleast 1 image in every mission plan ? ( Because [MAV_CMD_DO_MOUNT_CONTROL](http://mavlink.org/messages/common#MAV_CMD_DO_MOUNT_CONTROL) is the Mission command to control a camera or antenna mount) ? Can't we add this only when `mission_item_impl.get_camera_action() != MissionItem::CameraAction::NONE` ?

--Shakthi

julianoes, 2017-09-07:
When you say you added two mission items, did you remove the existing one from https://github.com/dronecore/DroneCore/blob/master/example/fly_mission/fly_mission_dronecore.cpp#L103-L108.

No, `MOUNT_CONTROL` should not have anything to do with taking pictures. It's for the gimbal only.

shakthi-prashanth-m, 2017-09-07:
Now, its not happening. May be My setup issue. (I re-installed gazebo for confirming it),
Thanks.

closed 2017-09-07 by shakthi-prashanth-m