mavlink/MAVSDK #61: add_XXX_connection() could be better named

hamishwillee, 2017-09-12:
Using connection here means that any description of the API is riddled with the word connection - since we also have to connect to the device. It is probably worth using something like  `add_XXX_port()` to make it clear that this is just part of the whole "connect to devices" process.

julianoes, 2018-07-23:
@JonasVautherin do you think that `add_XXX_port()` would be better?

JonasVautherin, 2018-08-06:
But it actually adds the connection, right? `add_udp_connection()` means "listen for the next mavlink broadcast you find using UDP and actually setup the connection to it". `add_udp_port()` may sound like "start listening on a new port", wouldn't it?

julianoes, 2018-08-07:
I mean actually it does start to listen and does not actually establish a connection to the drone as such.

JonasVautherin, 2018-08-07:
Couldn't we split it into `start_XXX_discovery()`, that would return information about the discovered drones, and `connect_XXX(...)` which would take that information and actually start the connection?

julianoes, 2018-08-07:
I guess but it's more complicated for an SDK user. `connect` is sort of the `auto System = dronecore.get_system()` call.

JonasVautherin, 2018-08-07:
Right, I see.

julianoes, 2018-08-17:
I vote we leave it as is.

closed 2018-08-17 by julianoes

hamishwillee, 2018-08-20:
Sure. This does not seem to be causing any confusion.