mavlink/MAVSDK #70: Allow system-wide installation

julianoes, 2017-09-14:
We should allow and support to install the library and headers into `/usr/local/`.

hamishwillee, 2017-09-14:
The docs say we already do :-) What about Windows?

julianoes, 2017-09-14:
Hm, let me try it.

julianoes, 2017-09-14:
You are right, that's already working.
```
sudo INSTALL_PREFIX=/usr/local make install
[sudo] password for julianoes: 
Build dir: build/default
[  3%] Built target gtest
[  8%] Built target gmock
[ 11%] Built target gtest_main
[ 64%] Built target dronecore
[ 72%] Built target unit_tests_runner
[100%] Built target integration_tests_runner
Install the project...
-- Install configuration: "Debug"
-- Installing: /usr/local/lib/libdronecore.a
-- Installing: /usr/local/include/dronecore/dronecore.h
-- Installing: /usr/local/include/dronecore/device.h
-- Installing: /usr/local/include/dronecore/device_plugin_container.h
-- Installing: /usr/local/include/dronecore/action.h
-- Installing: /usr/local/include/dronecore/info.h
-- Installing: /usr/local/include/dronecore/logging.h
-- Installing: /usr/local/include/dronecore/mission.h
-- Installing: /usr/local/include/dronecore/mission_item.h
-- Installing: /usr/local/include/dronecore/offboard.h
-- Installing: /usr/local/include/dronecore/telemetry.h
```

What about making it the default?

julianoes, 2017-09-14:
> What about making it the default?

The problem with that is that it breaks the `make docs`.

I'm wondering if we should get away of all the Makefile magic and instead just do cmake only.

closed 2017-09-14 by julianoes

hamishwillee, 2017-09-14:
@julianoes 
> What about making it the default?

That would be great, but only if it worked on Windows.

>> The problem with that is that it breaks the `make docs`.
> I'm wondering if we should get away of all the Makefile magic and instead just do cmake only.

I quite like having simple instructions for doing this stuff. At some point it would be good to auto-build the docs and auto-copy them into /docs/api-reference repo - presumably this can be done just as easily without make integration though.

hamishwillee, 2017-09-14:
PS. The bugger of this is though, that it would be much better for examples and everything else to have system install as the default or at least as the default documented. The CMakeLists.txt could be a lot simpler for a start. But since things don't work on Windows, we're kind of stuck with this as is.

julianoes, 2017-09-14:
> I quite like having simple instructions for doing this stuff. At some point it would be good to auto-build the docs and auto-copy them into /docs/api-reference repo - presumably this can be done just as easily without make integration though.

We can always just make a script that does it like `create_docs.sh`.

hamishwillee, 2017-09-15:
@julianoes 
Thinking about this, linux/mac is definitely the main OS and system install is the desired pattern. We should aim to support that as the default.

The problems are that:
1. It doesn't work on windows
   * ... and there is no way to tell that as a user
   * therefore users have to use local install and we have to have complicated CMakeLists.txt files.
2. It breaks doc builds
   - is this because doxygen is looking in the wrong folder locations?
   - not that doc build doesn't work on windows anyway right now. 

So, suggestion:
1. Make system install the default
   - If someone does this build on windows fail and display a message "System install not supported on windows - do `THE COMMAND to build locally`
2. Make docs build against system install path (doxygen config file change). On Windows fail the docs call with a message that"Automated docs builds not supported on Windows".
3. Example code has to stay the same - that should build on windows too.
4. Docs change to separate out how you can point to local installs as a separate topic.
Thoughts?

julianoes, 2017-09-15:
Ok, I support the suggestion.

hamishwillee, 2017-09-15:
@julianoes Awseome - do you need me to create an issue report?

julianoes, 2017-09-15:
Done in #73.