mavlink/MAVSDK #73: Make system-wide install default

julianoes, 2017-09-15:
As by @hamishwillee's suggestions in #70:

1. Make system install the default
   - If someone does this build on windows fail and display a message "System install not supported on windows - do `THE COMMAND to build locally`
2. Make docs build against system install path (doxygen config file change). On Windows fail the docs call with a message that"Automated docs builds not supported on Windows".
3. Example code has to stay the same - that should build on windows too.
4. Docs change to separate out how you can point to local installs as a separate topic.

julianoes, 2017-11-27:
This is done in #73 for Linux. Windows remains local because that's how it is generally done in the Windows world.

closed 2017-11-27 by julianoes

hamishwillee, 2017-11-27:
Understand your point, but this does make our docs quite complicated :-(

julianoes, 2017-11-28:
As far as I read up on it for Windows you add dependencies locally. The global install location would be C:\Windows\system32 but we all remember the dll hell that produced and it doesn't seem to be recommended anymore.

hamishwillee, 2017-11-28:
So where do people install shared binaries? More specifically, what will we do if we supply a binary release of DroneCore for Windows?

julianoes, 2017-11-28:
It gets included in the exe or if linked dynamically supplied in the folder with the exe, I think.

hamishwillee, 2018-04-11:
@Stifael Re https://github.com/dronecore/DroneCore/issues/73#issuecomment-380083682

The documentation is correct and up to date according to our process. The master reflects the current released version (0.1) which is VERY old. The develop reflects the head revision. When we make a release the master in both source code and docs will be updated to the current release version.

The problem here is that we haven't made a release forever, and it is easy to read the master instructions, which have very little to do with how people actually use the system (we encourage everyone to use develop). 

Upshot, if you use the develop docs and develop branch the docs definitely work. The master docs and master branch _should_ work too - but I have not made too much effort to backport docs because a new release has been imminent for ... ever.