mavlink/MAVSDK #84: Offboard mode can lead to crash

julianoes, 2017-09-21:
The offboard functionality needs some testing and bugfixing because it has lead to crashes in the past.

It seems to work fine when you start in NED mode first but can lead to a crash if started in body mode.

shakthi-prashanth-m, 2017-10-11:
@julianoes : I didn't observe crash in [this](https://github.com/dronecore/DroneCore/blob/develop/integration_tests/offboard_velocity.cpp#L92) case. 
I created a [sample app](https://gist.github.com/shakthi-prashanth-m/63e165fa5612dc36d55f34c764987999) to test offboard mode (using the code from integration_tests) with both NED & BODY mode.
Neither of them crashed. Any idea what caused crash ?

shakthi-prashanth-m, 2017-10-11:
```
$ DroneCore/example/offboard_velocity/build$ ./offboard NED
[06:46:44|Info ] New device on: 127.0.0.1:14557 (udp_connection.cpp:212)
In Air
sent velocity ned once
Offboard started
Let yaw settle...
sent velocity ned 628 times
sent velocity ned 400 times
sent velocity ned 400 times
sent velocity ned 400 times
sent velocity ned 400 times
Offboard stopped
Landed
Disarm failed:Command denied, not landed
-------------------------------------------------------------------------------
$ DroneCore/example/offboard_velocity/build$ ./offboard BODY
[06:47:34|Info ] New device on: 127.0.0.1:14557 (udp_connection.cpp:212)
In Air
sent velocity ned once
Offboard started
Turn around yaw & climb
Turn back
Wait for a bit
Fly a circle
Wait for a bit
Fly a circle sideways...
Wait for a bit
Offboard stopped
Landed
Disarmed
```

julianoes, 2017-10-12:
@shakthi-prashanth-m thank you. Could you make a pull request and add this test either as an example and/or as an integration test?

I will run this also against an internal version of Firmware to see if that's where the bug comes from.

shakthi-prashanth-m, 2017-10-13:
Sure @julianoes. I added [it](https://github.com/dronecore/DroneCore/pull/116) as an example. Please check.

julianoes, 2017-10-17:
This was a bug in our internal version of Firmware.

It was fixed through:
https://github.com/PX4/Firmware/commit/c41185a2771a3e791404043cdf107da57d996e2d
https://github.com/PX4/Firmware/commit/d9cec600edfe030bfea65900840935eb62e2f15b

closed 2017-10-17 by julianoes

shakthi-prashanth-m, 2017-10-17:
Cool :blush: Thanks @julianoes