mavlink/MAVSDK #91: Release builds fail

hamishwillee, 2017-09-26:
I build liked this:
```cd DroneCore
make clean
BUILD_TYPE=Release make
```
But when I run this  (or `BUILD_TYPE=Release make default`) I get lots of build errors like:
```/home/ubuntu/DroneCore/core/integration_test_helper.h: In member function 'virtual void SitlTest::SetUp()':
/home/ubuntu/DroneCore/core/integration_test_helper.h:12:38: error: ignoring return value of 'int system(const char*)', declared with attribute warn_unused_result [-Werror=unused-result]
         system("./start_px4_sitl.sh");
                                      ^
```

julianoes, 2017-09-26:
I can't reproduce this with current master (a30010bee275b6996f5bbfe3b4dbb8ab0792236a), neither on Linux, nor on macOS.

hamishwillee, 2017-09-27:
I just rebuilt current master on ubuntu Linxux and had same problem. Does this [gist help](https://gist.github.com/hamishwillee/9b4a6a282b25eb2429424169bf39db54). If not, should I perhaps try a completely fresh ubuntu?

julianoes, 2017-09-27:
Thank you, that helps because I can see:

```
-- The C compiler identification is GNU 6.2.0
-- The CXX compiler identification is GNU 6.2.0
```

Ok, I could not see these warnings with GCC 7.2 in Fedora 26 but I can reproduce them in Ubuntu 16.04 in docker actually.

I'll add the release builds to CI and then fix it.

hamishwillee, 2017-09-27:
Awesome! I'm using that GCC because it is the toolchain used to build PX4 for NuttX.

julianoes, 2017-10-12:
Fixed in #105.

closed 2017-10-12 by julianoes