mavlink/MAVSDK #100: Add support for NED missions

hamishwillee, 2017-09-27:
Generally when writing "reusable" drone applications and examples it is much better to be able to specify positions in NED co-ordinates rather than absolute positioning - because then you can specify a pattern of movement and re-use it as needed.

NED missions are supported by MAVLink, but not PX4. The proposal is to add a sensible API in DroneCore with a backend that can be replaced by PX4 support when that comes available.

dagar, 2017-09-27:
The PX4 side might not be that far off. https://github.com/PX4/Firmware/pull/7432

hamishwillee, 2018-02-12:
@shakthi-prashanth-m So caveat here, is that I don't know anything much about "NED" missions in mavlink other than that apparently they exist. We'd have to start by looking at what MAVLink expects to supply and work out our API from there. 

That said, I am *assuming* NED waypoints would be just like NED goto commands - relative to the home position OR to the current vehicle position rather than a GPS position. The API would look something like:
```
add_waypoint(north_m, east_m, down_m)
```
I suspect that the mission would define the frame globally (ie relative to home or current vehicle body), though depending on what PX4/MAVLink supports, it might be possible to supply a different frame for each mission item.

shakthi-prashanth-m, 2018-02-20:
@hamishwillee I hope you didn't mean Offboard position control. Did you ?