mavlink/MAVSDK #106: Dynamic plug-in infrastructure

shakthi-prashanth-m, 2017-09-28:
DroneCore has its functionality built with compile-time plug-ins such as `Action`, `Mission`, `Telemetry`, etc. When DroneCore is built, we can choose which plug-ins we're interested in. Build chooses plug-ins in `plugins` directory and generates `DevicePluginContainer` class and methods to access to the "choosen" plug-in objects. It is necessary to build DroneCore if we wanted to add a new plug-in to it.

We're interested to add new plug-ins which DroneCore will pick it up run-time. Theoretical approach for this is [here](https://px4.slack.com/archives/C68J8H32A/p1506435370000733).

However, as we're interested to developing language bindings using gRPC; gRPC requires `.proto` files for each of the plug-in, it is necessary for gRPC server to know them at compile-tim.
We're yet to investigate the feasibility of both (gRPC & dynamic plug-in). 

Updates on this activity will posted here.

julianoes, 2017-10-20:
One idea was to mangle multiple files together into one .proto. The plugins' proto files would basically be incomplete proto files that only work once embedded in a parent dronecore.proto file.

hamishwillee, 2017-10-22:
> One idea was to mangle multiple files together into one .proto. The plugins' proto files would basically be incomplete proto files that only work once embedded in a parent dronecore.proto file.

Would that allow run-time plugins (".proto files for each of the plug-in to be known at compile-time.")?

julianoes, 2017-10-23:
> Would that allow run-time plugins (".proto files for each of the plug-in to be known at compile-time.")?

The `.proto` files are used to generate the gRPC headers, so that would be compile time.

hamishwillee, 2017-10-23:
>> Would that allow run-time plugins (".proto files for each of the plug-in to be known at compile-time.")?

> The .proto files are used to generate the gRPC headers, so that would be compile time.

So interesting idea, but doesn't fit the original requirement.

julianoes, 2017-10-24:
Well, we can't have it all I think.

hamishwillee, 2017-10-24:
> Well, we can't have it all I think.

The whole problem to be addressed by this issue is summarised up the top as: "It is necessary to build DroneCore if we wanted to add a new plug-in to it."

So I'm not concerned about having it all, but the fact that we're talking about solutions that don't meet any stated requirement.

julianoes, 2017-10-25:
@hamishwillee so actually @avinash-palleti made the proposal in the last SDK call to use separate .proto files. This way would support plugins to be loaded at runtime.

Here are some notes:
https://docs.google.com/document/d/1qmmbpal6k8QYL2jGSRdpB1owIw9LU56EREuezfQbzN4/edit

hamishwillee, 2017-10-25:
Thanks! Works for me.

shakthi-prashanth-m, 2018-01-10:
#207 has same purpose. Closing this.

closed 2018-01-10 by shakthi-prashanth-m