mavlink/MAVSDK #112: [Plugin] RTK RTCM packets to FCU

AlexisTM, 2017-10-03:
Adding a plugin which can send the RTCM updates of the base station to the UAVs. Thus replacing the need for QGroundcontrol for in-flight operation.

julianoes, 2017-10-05:
Sounds like a good thing to have. Thanks for listing this.

mrpollo, 2021-02-10:
For context, there's some information on a duplicate issue #534

coderkalyan, 2021-02-11:
The most time consuming thing to develop here is probably a parser for GPS messages I think?

coderkalyan, 2021-02-11:
Or... not, https://github.com/PX4/PX4-GPSDrivers this exists apparently

AlexisTM, 2021-02-11:
The [rtk_ros](https://github.com/AlexisTM/rtk_ros) actually uses -an outdated version of- [PX4/GPSDrivers](https://github.com/PX4/PX4-GPSDrivers). We were using Mavros for that RTCM forward feature.

Katawann, 2021-02-11:
And from this issue #1252, this is how QGC is handle it:

> This is the functionality in QGC.
> https://github.com/mavlink/qgroundcontrol/blob/master/src/GPS/RTCM/RTCMMavlink.cc#L24
> 
> It is sending GPS_RTCM_DATA.
> https://mavlink.io/en/messages/common.html#GPS_RTCM_DATA

PR #1761 (2022-05-05): Add RTK plugin
First try without testing at sending RTCM correction data.

Closes #112.
closed 2022-05-05 by julianoes