mavlink/MAVSDK #121: Example Fly mission uses hard-coding of mission points

shakthi-prashanth-m, 2017-10-19:
Fly mission example uses[ hard-coding](https://github.com/dronecore/DroneCore/blob/develop/example/fly_mission/fly_mission.cpp#L73) of waypoints. It is hard for someone to compute longitude, latitude, etc and use this example.
I prefer, we can make this example take QGC Mission plan as input; where one can plan missions using QGC and feed its path via command-line to this example.

[image]

We can parse QGC mission plan and pass mission items to DroneCore Mission plugin.
Make sense ?

Regards,
Shakthi

julianoes, 2017-10-19:
Thanks for the issue. This has been requested several times actually and it would be a good thing to have.

The only catch is that not all mission items are currently supported in DroneCore, so we won't be fully compatible. We would have to just ignore unsupported items and warn the user about it.

hamishwillee, 2017-10-19:
@julianoes @shakthi-prashanth-m 

> Fly mission example uses hard-coding of waypoints. It is hard for someone to compute longitude, latitude, etc and use this example.

The solution to this problem is NED based mission points - as per #100.
I really think it makes sense to implement this ourselves, sooner rather than later.

> The only catch is that not all mission items are currently supported in DroneCore, so we won't be fully compatible. We would have to just ignore unsupported items and warn the user about it.

My take on this is that mission import should completely bypass whatever we do with the API. It would be something to add flexibility over the top of our existing API. So perhaps:
- Allow upload and download of files *or strings* in generic waypoint format. Users can then dynamically construct the messages if they want.
- Provide an API to send a generic waypoint message or custom message - so the user can send whatever message they like. 

Yes I know it removes some of the "clean-ness" of the API, but if we don't do this then I personally as a user would be forced to write my own plugin to do this.

shakthi-prashanth-m, 2018-02-11:
#235

hamishwillee, 2018-02-11:
@shakthi-prashanth-m So what does https://github.com/dronecore/DroneCore/issues/121#issuecomment-364714307 add to this discussion? 

I still think we should implement #100

shakthi-prashanth-m, 2018-02-12:
> @shakthi-prashanth-m So what does #121 (comment) add to this discussion?

@hamishwillee, sorry which comment ? The link you referred says that **you opened this on Oct 20, 2017**! :-)

>I still think we should implement #100

Yeah, I think this is useful too. So I [added my question](https://github.com/dronecore/DroneCore/issues/100#issuecomment-364835706) regarding NED mission for my understanding.

hamishwillee, 2018-02-12:
@shakthi-prashanth-m I was asking about why you added #235 to this discussion. I've responded re NED missions.

shakthi-prashanth-m, 2018-02-14:
Yeah, because this issue was about complexity of user's effort in uploading mission with hard-coding waypoint alt, lon. As #235 deals this indirectly, I referenced it.

julianoes, 2026-10-24:
The examples are usable, closing.

closed 2026-10-24 by julianoes