mavlink/MAVSDK #122: Add position control to offboard

julianoes, 2017-10-19:
Offboard currently only supports velocity inputs, however, it would be good to also have position control.

hamishwillee, 2017-10-19:
Yes! I think we need to consider whether we can make this simpler than velocity control. For example, should DC handle the resending of messages? I would say yes. 

It might also be nice to have an API that completes when you reach the destination.

julianoes, 2018-02-22:
Being addressed by #260.

afwilkin, 2020-03-18:
Hello, I was looking for something similar to this (a method like set_velocity_body()) under offboard mode. Is this still being developed?

julianoes, 2020-03-18:
This is done.

See: https://github.com/mavlink/MAVSDK/blob/7d7579659f8c12e3d9406a17eac96f3382153a5b/src/plugins/offboard/include/plugins/offboard/offboard.h#L208-L213

closed 2020-03-18 by julianoes

hamishwillee, 2020-03-19:
And [here in the docs](https://mavsdk.mavlink.io/develop/en/api_reference/classmavsdk_1_1_offboard.html#classmavsdk_1_1_offboard_1a0bcbc79ba1277f0e7ad110f86b902cc7)

Iko-Git, 2020-11-19:
Could you please give me a guideline for using _float mavsdk::Offboard::VelocityBodyYawspeed::forward_m_s {}_ into _ROS_ @hamishwillee @julianoes

julianoes, 2020-11-19:
@Iko-Git please don't necrobump and create a new issue.