mavlink/MAVSDK #127: Research MAVLInk statustexts

hamishwillee, 2017-10-22:
https://github.com/dronecore/DroneCore/pull/119 adds support to print statustexts from the vehicle. 

This issue is to review the messages and determine whether they:
- Can remain FYI console logs
- Should be escalated to users
- Should result in code errors
- Require other actions. 

Also determine best place for handling messages (proposal is to defer to relevant plugins). 

Relevant discussion:  https://github.com/dronecore/DroneCore/pull/119#pullrequestreview-69720267

JonasVautherin, 2019-06-11:
@julianoes is that solved now that we have the statustext in the telemetry plugin?

julianoes, 2019-06-11:
This will improve once we have the event interface, closing for now.

closed 2019-06-11 by julianoes