mavlink/MAVSDK #136: Review MAVLink messages and determine appropriate actions

hamishwillee, 2017-10-26:
@julianoes @mrpollo As previously discussed, we should assess mavlink messages and determine if we actually need to take any action when we recieve any of them.
1. Below are listed the ones that come out of our examples. The only one that looks at all interesting is the fact that the system considers it a critical issue that default/minimum takeoff height is used.
2. @julianoes do you think we could change the format from "mavlink info" to "MAVLink info:" (etc.)

```
[07:39:06|Debug] mavlink info: [cmd] Mission #1 loaded, 6 WPs, curr: 0 (device_impl.cpp:223)
[07:39:06|Debug] mavlink info: [logger] file: rootfs/fs/microsd/log/2017-10-26/0 (device_impl.cpp:223)
[07:39:23|Debug] mavlink info: ARMED by arm/disarm component command (device_impl.cpp:223)
In Air...
[07:39:23|Debug] mavlink info: Using minimum takeoff altitude: 2.50 m (device_impl.cpp:223)
[07:39:23|Debug] mavlink info: Takeoff detected (device_impl.cpp:223)
> [07:39:24|Debug] mavlink critical: Using minimum takeoff altitude: 2.50 m (device_impl.cpp:223)
[07:40:17|Debug] mavlink info: Landing at current position (device_impl.cpp:223)
[07:44:12|Debug] mavlink info: data link #0 lost (device_impl.cpp:223)
[07:44:12|Debug] mavlink info: data link #0 regained (device_impl.cpp:223)
[07:44:13|Debug] mavlink info: Executing mission. (device_impl.cpp:223)
[07:44:13|Debug] mavlink info: Takeoff to 2.5 meters above home. (device_impl.cpp:223)
[07:44:13|Debug] mavlink info: Takeoff detected (device_impl.cpp:223)
[07:44:13|Debug] mavlink info: Takeoff to 10.0 meters above home. (device_impl.cpp:223)
[07:45:06|Debug] mavlink info: Mission finished, loitering. (device_impl.cpp:223)
[07:45:06|Debug] mavlink info: RTL: climb to 518 m (30 m above home) (device_impl.cpp:223)
[07:45:14|Debug] mavlink info: RTL: return at 518 m (30 m above home) (device_impl.cpp:223)
[07:45:19|Debug] mavlink info: RTL: descend to 493 m (5 m above home) (device_impl.cpp:223)
[07:45:44|Debug] mavlink info: RTL: loiter 5.0s (device_impl.cpp:223)
[07:45:49|Debug] mavlink info: RTL: land at home (device_impl.cpp:223)
[07:46:00|Debug] mavlink info: Landing detected (device_impl.cpp:223)
[07:46:03|Debug] mavlink info: DISARMED by auto disarm on land (device_impl.cpp:223)
```

hamishwillee, 2017-10-31:
A few more
```
[05:58:02|Debug] MAVLink: emergency: Accel #0 fail:  TOUT! (device_impl.cpp:223)
[05:58:02|Debug] MAVLink: emergency: Gyro #0 fail:  TOUT! (device_impl.cpp:223)
```

julianoes, 2026-09-15:
Capitalization done, and clearly the status quo is ok to print out the statustext messages is ok - otherwise there would be more complaints.

closed 2026-09-15 by julianoes