mavlink/MAVSDK #148: Add support for MAVLink communication faking/mocking

hamishwillee, 2017-11-02:
In order to allow sensible unit tests to be written.

shakthi-prashanth-m, 2017-11-04:
@hamishwillee @julianoes Can you please describe more about this task ?

hamishwillee, 2017-11-05:
Unit tests are supposed to test just the functionality of a particular component. As each plugin is a MAVLink processor - sending and/or receiving MAVLink messages to unit test our code we would need to  create some  mock system that can sensibly act as a source/sink of MAVLink and respond to messages from our plugins as "required".

This would allow us  to unit test our code in isolation from other plugins and with completely repeatable behaviour, on any platform.

For our integration tests we use SITL as a sync/source but that is no good for unit tests, because SITL requires all sort of setup, and does not respond repeatably on different systems, or even the same system on a different load. 

I have no idea what "exactly" this would look like, but I'd start by considering the QGroundControl mocklink.

julianoes, 2017-11-06:
Yes it could be something like a `FakeConnection` that whenever you send it a command it that is correct it would respond with ack. Additionally you could also check the failure cases such as timeouts, or responding with denied if a param doesn't make sense.