mavlink/MAVSDK #151: External library may not work and is not providing errors

hamishwillee, 2017-11-05:
I tried to build the external library on linux (develop branch) but there is no way to tell if it worked, and I think it needs some more error checking.

```		
make EXTERNAL_DIR=external_example		
```
- Did install, but the example plugin header is not exported
- Tried to run the external hello world integration test as per:
   ```		
   build/default/integration_tests_runner --gtest_filter="ExternalExampleHello"		
   ```
   But got no tests run.
- Tried to list all tests using the gtest tricks instructions. The above test was not listed.
- Tired to build using an invalid directory - build "succeeded"
```		
make EXTERNAL_DIR=impossible_external_example		
```

So, some fixes needed:
1. Should tell you if you try to build a directory that does not exist (and fail build)
2. Should tell you if you try to run a test that is not defined
3. Above should work :-)
4. If I moved external_example directory to same level as DroneCore folder, would it be added like this (as you would expect)
```
make EXTERNAL_DIR=../external_example
```

hamishwillee, 2017-11-06:
OK, so I 'll update docs with `make clean`. FYI, I tried this (latest develop branch) and got the error on `make`:
```
External plugin folder: 
Found plugin /home/ubuntu/DroneCore/plugins/action
Found plugin /home/ubuntu/DroneCore/plugins/gimbal
Found plugin /home/ubuntu/DroneCore/plugins/info
Found plugin /home/ubuntu/DroneCore/plugins/logging
Found plugin /home/ubuntu/DroneCore/plugins/mission
Found plugin /home/ubuntu/DroneCore/plugins/offboard
Found plugin /home/ubuntu/DroneCore/plugins/telemetry
done: 
CMake Error at CMakeLists.txt:158 (add_subdirectory):
  The source directory

    /home/ubuntu/DroneCore/libs/tinyxml2

  does not contain a CMakeLists.txt file.

-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
-- GRPC_DIR not set, checking automatically for gRPC
-- gRPC not found, skipping gRPC build
-- Configuring incomplete, errors occurred!
See also "/home/ubuntu/DroneCore/build/default/CMakeFiles/CMakeOutput.log".
See also "/home/ubuntu/DroneCore/build/default/CMakeFiles/CMakeError.log".
Build dir: build/default
/bin/sh: 1: cd: can't cd to build/default
Makefile:71: recipe for target 'default' failed
make: *** [default] Error 2
```

julianoes, 2017-11-07:
Yer, that's leftover folders. You can clean it up doing `git clean -dfx` but make sure to check first with `git clean -dfxn` what you're about to delete.

hamishwillee, 2017-11-07:
It may be that all the fixes needed are not yet in develop. After syncing to develop and running above I still get errors:
```
[ 62%] Linking CXX shared library libdronecore.so
/usr/bin/ld: cannot find -ltinyxml2
collect2: error: ld returned 1 exit status
CMakeFiles/dronecore.dir/build.make:1004: recipe for target 'libdronecore.so' failed
make[3]: *** [libdronecore.so] Error 1
CMakeFiles/Makefile2:179: recipe for target 'CMakeFiles/dronecore.dir/all' failed
make[2]: *** [CMakeFiles/dronecore.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make[1]: *** [all] Error 2
Makefile:71: recipe for target 'default' failed
make: *** [default] Error 2
```

julianoes, 2017-11-08:
Ok, there is a new dependency. On linux you need libtinyxml2.

Therefore, on Ubuntu: `sudo apt-get install libtinyxml2-dev`

hamishwillee, 2017-11-08:
Thanks. Agree fixed. Closing now.

closed 2017-11-08 by hamishwillee