mavlink/MAVSDK #157: Example fails with library not found after #130

hamishwillee, 2017-11-09:
Following #130 takeoff_and_land example fails to find the library at runtime:
```
failed to run with error `./takeoff_and_land: error while loading shared libraries: libdronecore.so: cannot open shared object file: No such file or directory`. 
```
The **libdronecore.so** file is present in /usr/local/libs/.

----
Full build log below
```
ubuntu@ubuntu:~/DroneCore/example/takeoff_land/build$ cmake ..
-- The C compiler identification is GNU 6.2.0
-- The CXX compiler identification is GNU 6.2.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ubuntu/DroneCore/example/takeoff_land/build
ubuntu@ubuntu:~/DroneCore/example/takeoff_land/build$ make -j4
Scanning dependencies of target takeoff_and_land
[ 50%] Building CXX object CMakeFiles/takeoff_and_land.dir/takeoff_and_land.cpp.o
[100%] Linking CXX executable takeoff_and_land
[100%] Built target takeoff_and_land
ubuntu@ubuntu:~/DroneCore/example/takeoff_land/build$ ./takeoff_and_land 
./takeoff_and_land: error while loading shared libraries: libdronecore.so: cannot open shared object file: No such file or directory
ubuntu@ubuntu:~/DroneCore/example/takeoff_land/build$ 
ubuntu@ubuntu:~/DroneCore/example/takeoff_land/build$ make -j4
[100%] Built target takeoff_and_land
ubuntu@ubuntu:~/DroneCore/example/takeoff_land/build$ ./takeoff_and_land 
./takeoff_and_land: error while loading shared libraries: libdronecore.so: cannot open shared object file: No such file or directory
ubuntu@ubuntu:~/DroneCore/example/takeoff_land/build$ 
```

julianoes, 2017-11-13:
The problem seems to be that `/usr/local/lib` is not in the `$LD_LIBRARY_PATH`.

You can workaround the problem using
```
export LD_LIBRARY_PATH="$LD_LIBRARY_PATH:/usr/local/lib"
```

I need to research how other libraries do this.

julianoes, 2017-11-13:
According to http://tldp.org/HOWTO/Program-Library-HOWTO/shared-libraries.html it is ok to install libraries into `/usr/local/lib` when you compile from source, so during development.

For Ubuntu, all that is required after `sudo make default install` is `sudo ldconfig` to update the linker cache.

hamishwillee, 2017-11-13:
Thanks very much. I'll add a note in the build instructions and a troubleshooting section in the examples section.

closed 2017-11-13 by hamishwillee

rt-2pm2, 2018-07-27:
I got the same error on my Fedora28. Setting the environment variable LD_LIBRARY_PATH didn't solve the problem. 
Also running the ldconfig didn't solve. I found that ldconfig looks into "/etc/ld.so.conf". Checking there I saw that there was no reference to those share libraries.
In the "etc/ld.so.conf" there is simply an include of the files *.conf contained in "etc/ld.so.conf.d/".

I added a new file called "dronecode.conf" containing the path to the libraries in that folder. Then, after running the ldconfig, it worked. 

p.s. The reference about the ldconfig: [](https://stackoverflow.com/questions/12045563/cannot-load-shared-library-that-exists-in-usr-local-lib-fedora-x64)

julianoes, 2018-07-30:
@rt-2pm2 can you please post the whole output? I'm doing this all the time in Fedora, so I have the feeling you're missing something somewhere.

rt-2pm2, 2018-07-30:
@julianoes Sure, I am happy to participate. 
To restore the system in the condition before my changes:
1) Remove the "dronecode.conf" file from the "etc/ld.so.conf.d" folder. In this way it won't be included by ldconfig.
2) Run the `ldconfig`

I check that the libraries are in "/usr/local/lib"
> $ ls /usr/local/lib                                                          
libdronecode_sdk_action.so       libdronecode_sdk_offboard.so
libdronecode_sdk_calibration.so  libdronecode_sdk.so
libdronecode_sdk_camera.so       libdronecode_sdk_telemetry.so
libdronecode_sdk_follow_me.so    
libdronecode_sdk_gimbal.so      
libdronecode_sdk_info.so        
libdronecode_sdk_logging.so      
libdronecode_sdk_mission.so     

Now I try to run one  of my example without using my approach:
The LD_LIBRARY_PATH environment variable in my fedora is void. Thus, I assign the environment variable with 
> $ LD_LIBRARY_PATH="/usr/local/lib"
I update the linker cache running `ldconfig`

Then, when I run the executable (fly_auto), I get this output:
> $ ./fly_auto udp://127.0.0.1:6000                                            
./fly_auto: error while loading shared libraries: libdronecode_sdk.so: cannot open shared object file: No such file or directory

I hope that I answered your request.

julianoes, 2018-07-31:
@rt-2pm2 ok please try with `export`:
```
export LD_LIBRARY_PATH="/usr/local/lib"
```

rt-2pm2, 2018-07-31:
@julianoes Yes, I tried and it works if I put the "export" keyword. 

My fault, maybe I misspelled the variable name when I tried the first time and, moreover, I didn't remember well the "export" effect: the application I run is not anymore the shell process, thus without "export" the variable will go out of scope. Thus, it is a viable solution to the problem, and including the variable in the .bashrc it will avoid me adding it every bash session.

Thanks

julianoes, 2018-07-31:
Exactly! Glad I could help!