mavlink/MAVSDK #162: Building example locally on windows fails after #130

hamishwillee, 2017-11-14:
I buil DroneCore locally on windows and installed it locally using latest instructions on master. I then updated the takeoff_and_land CMakeLists.txt for local install and built it.

There are a bunch of errors: https://gist.github.com/hamishwillee/eea2b8d9699fda1ff5c09bc79e4554e5

julianoes, 2017-11-14:
```
  ..\..\..\install\lib\dronecore.lib : warning LNK4272: library machine type 'x64' conflicts with target machine type 'x86' [D:\Github\delmet
emp\DroneCore\example\takeoff_land\build\takeoff_and_land.vcxproj]
```

I think you need to always specify x64 for all parts that you are building.

hamishwillee, 2017-11-14:
How? I'm using the bog standard instructions for cmake on Windows - at not point do I say x64 or x86.

julianoes, 2017-11-15:
You need to add `-G "Visual Studio 15 2017 Win64"` like here https://docs.dronecore.io/en/contributing/build.html#windows

hamishwillee, 2017-11-16:
I am starting to hate windows. For whatever reason VS is not properly installing paths in the console apps, so this now all isn't building. I am going to add that above and perhaps consider this done. It's that or murder someone.

hamishwillee, 2017-11-16:
OK, so basically that fixed one link problem, but still getting lots of unresolved externals::
```
  dronecore.lib(tcp_connection.obj) : error LNK2019: unresolved external symbol __imp_connect referenced in function "private: enum
dronecore::DroneCore::ConnectionResult __cdecl dronecore::TcpConnection::setup_port(void)" (?setup_port@TcpConnection@dronecore@@AEA
A?AW4ConnectionResult@DroneCore@2@XZ) [D:\Github\delmetemp\DroneCore\example\takeoff_land\build\takeoff_and_land.vcxproj]
  dronecore.lib(tcp_connection.obj) : error LNK2019: unresolved external symbol __imp_inet_addr referenced in function "private: enu
m dronecore::DroneCore::ConnectionResult __cdecl dronecore::TcpConnection::setup_port(void)" (?setup_port@TcpConnection@dronecore@@A
EAA?AW4ConnectionResult@DroneCore@2@XZ) [D:\Github\delmetemp\DroneCore\example\takeoff_land\build\takeoff_and_land.vcxproj]
  dronecore.lib(tcp_connection.obj) : error LNK2019: unresolved external symbol __imp_recv referenced in function "private: static v
oid __cdecl dronecore::TcpConnection::receive(class dronecore::TcpConnection *)" (?receive@TcpConnection@dronecore@@CAXPEAV12@@Z) [D
:\Github\delmetemp\DroneCore\example\takeoff_land\build\takeoff_and_land.vcxproj]
  D:\Github\delmetemp\DroneCore\example\takeoff_land\build\Debug\takeoff_and_land.exe : fatal error LNK1120: 17 unresolved externals
 [D:\Github\delmetemp\DroneCore\example\takeoff_land\build\takeoff_and_land.vcxproj]
```
There were some warnings coming out of the end of the DroneCore build too. 
"Most of full log" for building dronecore then takeoff_and_land here:
 https://gist.github.com/hamishwillee/eea2b8d9699fda1ff5c09bc79e4554e5

julianoes, 2017-11-16:
I'll look into it when I have leftover time :D .

hamishwillee, 2017-11-16:
See you next year then :-) It is odd that all the unresolved externals are in the connection classes.

julianoes, 2017-11-17:
@hamishwillee these are actually the TCP connection methods which are not available on Windows, so presumably we need to fix the linking there.

hamishwillee, 2017-11-20:
Ah, that makes sense!  @Rjasuja might have some thoughts on this given was original contributor of that functionality.

hamishwillee, 2017-11-28:
Docs updated in https://github.com/dronecore/docs/pull/71

PR #178 (2017-11-28): Fix Windows linking
This should fix the linking on Windows. We now properly link the static library itself to ws2_32.lib.

Also, the examples are now built in CI.

Closes #162.
closed 2017-11-28 by julianoes