mavlink/MAVSDK #176: Serial connection at 100% CPU

bertbuchholz, 2017-11-27:
When using a Pixhawk connected via a serial connection (dronecore::DroneCore::add_serial_connection()), the CPU usage for one core goes to 100%. Looking at it with a profiler, I found that SerialConnection::receive() has a while loop that reads as fast as possible and the read() used is on a non-blocking file descriptor, leading to this behavior.

On embedded devices, having 100% CPU usage is not feasible. I tried solving the issue by removing the O_NONBLOCK option when opening the file descriptor, but that didn't work (i.e., it behaved the same way). Then I added small sleeps into the receive method which works, but feels more like a workaround.

```
void SerialConnection::receive(SerialConnection *parent)
{
    // Enough for MTU 1500 bytes.
    char buffer[2048];

    const std::chrono::milliseconds wait_time_after_read(5);

    while (!parent->_should_exit) {
        int recv_len = read(parent->_fd, buffer, sizeof(buffer));
        if (recv_len < -1) {
            LogErr() << "read failure: " << GET_ERROR(errno);
        }
        if (recv_len > static_cast<int>(sizeof(buffer)) || recv_len == 0) {
            std::this_thread::sleep_for(wait_time_after_read);
            continue;
        }
        parent->_mavlink_receiver->set_new_datagram(buffer, recv_len);
        // Parse all mavlink messages in one data packet. Once exhausted, we'll exit while.
        while (parent->_mavlink_receiver->parse_message()) {
            parent->receive_message(parent->_mavlink_receiver->get_last_message());
        }
        std::this_thread::sleep_for(wait_time_after_read);
    }
}  
```

The 5ms in the example are quite a lot, 1ms also reduces the CPU usage a lot already. So while this works, it seems like not a proper solution. I wonder if the unlimited loop speed is by design or if it's a bug and should/will be fixed.

shakthi-prashanth-m, 2017-11-27:
@bertbuchholz , can you please try adding `O_NDELAY` instead of `O_NONBLOCK` [here](https://github.com/dronecore/DroneCore/blob/develop/core/serial_connection.cpp#L69) ?

julianoes, 2017-11-27:
@bertbuchholz thanks for reporting this.

The port is clearly opened in the wrong way. Since, we read in a separate thread, we can do blocking reads.

It would be great if you could test #177.

bertbuchholz, 2017-11-27:
I tried the fix in #177 and also @shakthi-prashanth-m's suggestion of using O_NDELAY instead of O_NBLOCKING but unfortunately, both still show the 100% CPU behavior.

So somehow, the port is still opened non-blocking or something makes the read() return immediately even if there's no data on it. In the #177 branch, I just added output to the condition:
```
if (recv_len > static_cast<int>(sizeof(buffer)) || recv_len == 0) {
            std::cout << "receive()" << recv_len;
            continue;
        }
```
and it's printing "0" all the time, so clearly it's still not blocking.

bertbuchholz, 2017-11-27:
I just realized, the reason it's still not blocking is that VMIN and VTIME for the termios settings are both zero. I don't know enough about this stuff to change this sensibly.

julianoes, 2017-11-27:
@bertbuchholz you are right, thanks for checking this. I've added a commit to fix `VMIN`.

julianoes, 2017-11-27:
@bertbuchholz I've merged this to `develop`. Please test again and re-open if you still have the issue. Thanks for the quick responses here.

closed 2017-11-27 by julianoes

bertbuchholz, 2017-11-28:
Yes, the VMIN setting does it, thanks for the fast fix!

PR #177 (2017-11-27): core: do blocking read on serial port
We should be able to read in a blocking way on the serial port since it
happens on a separate thread.

Fixes #176.