mavlink/MAVSDK #180: Few useful doxygen tags in the source leads to Doc build error

shakthi-prashanth-m, 2017-11-28:
Adding doxygen tags `@sa, @note, @para` leads to below error during `./generate_doc.sh`
```
  Generating: /home/intel/karma/fork/follow-me/fresh/DroneCore/install/docs/xml/classdronecore_1_1_follow_me.xml (class xml)
Unhanded kind in simplesect tag in markdown_any_tag:kind: see
Traceback (most recent call last):
  File "/home/intel/karma/fork/follow-me/fresh/DroneCore/generate_markdown_from_doxygen_xml.py", line 968, in <module>
    markdown_string=currentClass.markdown()
  File "/home/intel/karma/fork/follow-me/fresh/DroneCore/generate_markdown_from_doxygen_xml.py", line 871, in markdown
    outputstring+=the_function.markdown()
  File "/home/intel/karma/fork/follow-me/fresh/DroneCore/generate_markdown_from_doxygen_xml.py", line 586, in markdown
    functionstring+='\n\n%s' % markdown_any_tag(self.detaileddescription).strip()
  File "/home/intel/karma/fork/follow-me/fresh/DroneCore/generate_markdown_from_doxygen_xml.py", line 112, in markdown_any_tag
    child_text += markdown_any_tag(child,html=html,para=para) #.strip()
  File "/home/intel/karma/fork/follow-me/fresh/DroneCore/generate_markdown_from_doxygen_xml.py", line 100, in markdown_any_tag
    child_text+=str(ET.tostring(child),'utf-8')
```
@hamishwillee Does it need fix in  `generate_markdown_from_doxygen_xml.py` ?

hamishwillee, 2017-11-29:
Have you any actual test code I can run this against - ie showing that we need this? the follow-me current tree doesn't appear to use it?

I'm very tempted by @note, possibly by @sa. Why @para though? You get para support automatically just by having a new line.

shakthi-prashanth-m, 2017-11-29:
@hamishwillee  yes I had the code using tags @note, @sa in the code. As the build failed I removed them. I can upload the code which has these tags usage, if you want to test.

hamishwillee, 2017-11-29:
Thanks @shakthi-prashanth-m , but I wrote some test examples. You can try out the new version in #182 if you like.

hamishwillee, 2017-11-29:
This should be fixed now in develop. Give it a whirl :-)

closed 2017-11-29 by hamishwillee