mavlink/MAVSDK #189: Follow Me has no example

hamishwillee, 2017-12-04:
It would be nice :-)

shakthi-prashanth-m, 2017-12-04:
Like I said [here](https://github.com/dronecore/DroneCore/pull/142#issuecomment-348882562),

hamishwillee, 2017-12-04:
Arguable. Of course it would be better, but you can write a useful example simulate data, as you have for the integration test. You can even simulate the sort of "behaviour" of the APIs if you were so inclined. Completely up to you.

shakthi-prashanth-m, 2017-12-04:
OK, we can make an example similar to [MultiLocation](https://github.com/dronecore/DroneCore/blob/develop/integration_tests/follow_me.cpp#L79) case in [integration_tests/follow_me.cpp](https://github.com/dronecore/DroneCore/blob/develop/integration_tests/follow_me.cpp). As we discussed sometime ago, we can use an algorithm to generate GPS locations (simulating Location services of the underlying platform)and feed it to FollowMe plugin. @julianoes , does this sound good ?

julianoes, 2017-12-04:
@shakthi-prashanth-m Sure, sounds good.

I would vote to make the example rather minimal, so you would not expose all of the config options but just give the most  obvious use case.

hamishwillee, 2017-12-04:
> I would vote to make the example rather minimal, so you would not expose all of the config options but just give the most obvious use case.

Yes, though it would be good if it could emulate "proper usage" of a location source. By which I mean that if the sources are normally asynchronous callbacks whenever new data arrives, it would be good to show how to write an app that works with that kind of source.

shakthi-prashanth-m, 2018-01-12:
Closing it as PR #217 added Follow Me example.

closed 2018-01-12 by shakthi-prashanth-m