mavlink/MAVSDK #190: Global installation not working on Ubuntu

hamishwillee, 2017-12-04:
I followed our standard instructions:
```
make clean
make default
sudo make default install
```
I get
```
ubuntu@ubuntu:~/DroneCore$ sudo make default install
Build dir: build/default
make[1]: *** No rule to make target 'install'.  Stop.
Makefile:71: recipe for target 'default' failed
make: *** [default] Error 2
```

shakthi-prashanth-m, 2017-12-04:
@hamishwillee , I didn't get it, I followed same steps. Did you build on latest **develop** ?
```
shakthi@ubuntu:~/fork/DroneCore$ sudo make default install
Build dir: build/default
[  2%] Built target gtest
[  6%] Built target gmock
[  9%] Built target gtest_main
[ 62%] Built target dronecore
[ 71%] Built target unit_tests_runner
[100%] Built target integration_tests_runner
Install the project...
-- Install configuration: "Debug"
-- Installing: /usr/local/lib/libdronecore.so
-- Up-to-date: /usr/local/include/dronecore/dronecore.h
-- Up-to-date: /usr/local/include/dronecore/device.h
-- Installing: /usr/local/include/dronecore/device_plugin_container.h
-- Up-to-date: /usr/local/include/dronecore/action.h
-- Up-to-date: /usr/local/include/dronecore/follow_me.h
-- Up-to-date: /usr/local/include/dronecore/gimbal.h
-- Up-to-date: /usr/local/include/dronecore/info.h
-- Up-to-date: /usr/local/include/dronecore/logging.h
-- Up-to-date: /usr/local/include/dronecore/mission.h
-- Up-to-date: /usr/local/include/dronecore/mission_item.h
-- Up-to-date: /usr/local/include/dronecore/offboard.h
-- Up-to-date: /usr/local/include/dronecore/telemetry.h
```

hamishwillee, 2017-12-04:
Damn it. I will try again tomorrow. Thanks.

hamishwillee, 2017-12-04:
Agreed! Sadly I had to reinstall on completely clean OS to get this to work - just cleaning the tree and creating a fresh clone of dronecore was not enough :-(

Thanks for your help.

closed 2017-12-04 by hamishwillee