mavlink/MAVSDK #191: FollowMe API feels inconsistent

hamishwillee, 2017-12-04:
* `bool set_config (const Config &config)` - I would expect this to return a `Result` type
* `void get_last_location(TargetLocation &last_location)` could reasonably be be `TargetLocation& get_last_location()`
* If we are checking ack, then `void  set_curr_target_location(const TargetLocation &location)` should return a `Result`.

shakthi-prashanth-m, 2017-12-04:
@hamishwillee Thanks for reporting this. I will make changes accordingly.

julianoes, 2017-12-04:
> There should usually be matching async APIs.

I have no strong opinion on this one. We can add async/sync APIs as required.

hamishwillee, 2017-12-05:
>> There should usually be matching async APIs.

> I have no strong opinion on this one. We can add async/sync APIs as required.

Fine by me. This was just a dump of all the things that seemed odd :-)

hamishwillee, 2017-12-14:
This will be fixed by https://github.com/dronecore/DroneCore/pull/199

shakthi-prashanth-m, 2017-12-18:
@hamishwillee shall we close this ?
I believe Follow Me example is not part of this issue. Isn't it ?

hamishwillee, 2017-12-18:
Yes, fixed :-)

closed 2017-12-18 by hamishwillee