mavlink/MAVSDK #192: Running examples fails to find path to libdronecore.so

shakthi-prashanth-m, 2017-12-06:
I did
```
$ make clean
$ sudo make default install
$ cd example/takeoff_land
$ mkdir build
$ cd build
$ cmake ..
$ make
$ ./takeoff_land => leads to runtime error
./takeoff_and_land: error while loading shared libraries: libdronecore.so: cannot open shared object file: No such file or directory
```

However, after I export `LD_LIBRARY_PATH`, it works well.
```
$ export LD_LIBRARY_PATH=/usr/local/lib/
$ ./takeoff_and_land 
Waiting to discover device...
[01:55:17|Info ] New device on: 127.0.0.1:14557 (udp_connection.cpp:211)
[01:55:17|Info ] Applying default FollowMe configuration FollowMe to the device... (follow_me_impl.cpp:182)
[01:55:17|Debug] MAVLink: info: data link #0 lost (device_impl.cpp:229)
[01:55:17|Debug] MAVLink: info: data link #0 regained (device_impl.cpp:229)
[01:55:18|Debug] Discovered 4294967298 (dronecore_impl.cpp:234)
Discovered device with UUID: 4294967298
...
```

I think we need to change `CMakeLists.txt` during installing dronecore library.
It seems [this](https://github.com/dronecore/DroneCore/blob/develop/CMakeLists.txt#L192) line in CMakeLists.txt doesn't have effect of exporting dronecore library path.

Shouldn't we instruct user to export this if not done already ?

hamishwillee, 2017-12-06:
I believe this is a Linux "thing". It is a bit rubbish.

What I did was add the following notes when building the library and when running examples:
- https://docs.dronecore.io/v/develop/en/contributing/build.html#dronecore_system_wide_install
- https://docs.dronecore.io/v/develop/en/examples/#linux-error-loading-shared-libraries

If this could happen automagically as part of doing system wide installation that would be even better.

shakthi-prashanth-m, 2017-12-07:
@hamishwillee  Thanks.

closed 2017-12-07 by shakthi-prashanth-m