mavlink/MAVSDK #193: Exposing Flight mode APIs in core/device_impl.h

shakthi-prashanth-m, 2017-12-06:
We have [APIs](https://github.com/dronecore/DroneCore/blob/develop/core/device_impl.h#L86-L89) in `dronecore::DeviceImpl` for PX4 param get/set; but we don't have APIs to change Flight mode from a plugin. As of now we're doing the hard way by using  `MAV_CMD_DO_SET_MODE`. I think it makes sense to expose an API for the same as part of `dronecore::DeviceImpl` interface. 

Below are the plugins each of which maps to a **PX4 flight mode**:
- [Offboard plugin](https://github.com/dronecore/DroneCore/blob/develop/plugins/offboard/offboard_impl.cpp#L48)
- [FollowMe plugin](https://github.com/dronecore/DroneCore/blob/develop/core/device_impl.h#L86-L89)
- [Mission plugin](https://github.com/dronecore/DroneCore/blob/develop/plugins/mission/mission_impl.cpp#L682)

It is intuitive to move [flight mode APIs in Telemetry plugin](https://github.com/dronecore/DroneCore/blob/develop/plugins/telemetry/telemetry.h#L387-L392) to `dronecore::DeviceImpl` as well.

hamishwillee, 2017-12-06:
Having this in the `DeviceImpl` for plugins makes sense to me. I would keep the public API in telemetry -
 that is where I would expect to look for it as an API user.keeping the "public" API in Telemetry

julianoes, 2017-12-07:
I agree that this would be handy.

hamishwillee, 2017-12-07:
So the answer is Yes. Feel free to close  - and add the functionality/or change this to an issue and keep open.

julianoes, 2017-12-07:
The answer is yes but I won't get to it just yet.

shakthi-prashanth-m, 2018-01-15:
#206 addressed it.

closed 2018-01-15 by shakthi-prashanth-m