mavlink/MAVSDK #200: Name change proposal for "Device" class to "Drone" or "Vehicle"

shakthi-prashanth-m, 2017-12-14:
This is regarding name change proposal of the class `Device`, `DeviceImpl`, etc.
We've named the library & API after **drone**, as in **DroneCore**, it makes sense to name its core service class to be `class Drone`, or at least `class Vehicle` (in QGroundControl) rather than `class Device`.

Cons of having `Device` name is that it is likely to be used for Ground Control Station, Mobile device also.
In other words, GCS is a device as well;
When we say `Vehicle`, there is room of confusion whether it is Drone or Ground Control Station.

If DroneCore is also intended for general Robotics application, then probably `Device` might sound good.
Kindly share your thoughts, opinions.

Thanks.

julianoes, 2017-12-14:
I chose device initially because DroneCore can also be used for a land robot or a (mavlink enabled) camera only.

hamishwillee, 2017-12-14:
Device works for me given the camera-support aspect - and indeed there is no reason we couldn't do antenna trackers or any other mavlink controlled thingy.

shakthi-prashanth-m, 2017-12-15:
Alright, if its used for standalone devices like camera, then Device is fine.
Thanks @julianoes @hamishwillee .

closed 2017-12-15 by shakthi-prashanth-m