mavlink/MAVSDK #201: TCP connection drop causes high CPU usage

fklabach, 2017-12-15:
I connected DroneCore to a TCP server using `add_tcp_connection()`. If the added TCP connection drops, DroneCore will just spin in `receive()` of `TcpConnection`, because `recv()` returns immediately with 0. This causes 100% CPU load and shouldn't be desired behavior when the socket is disconnected.

I've fixed the issue by just setting `_should_exit` flag when `recv()` returns with 0 in receive thread. Certainly not the best solution but does the job for me. The dead connection still stays in DroneCore's connection list, so it would be nice if it could either be reopened manually through API or DroneCore tries to keep up connections itself.

julianoes, 2017-12-17:
@fklabach thanks for the report.

When a TCP connection is dropped, can you recover it? Or would you need to disconnect and connect again? I'm wondering on what level this should be fixed.

fklabach, 2017-12-17:
I'm not sure if I get you right, I can recover the physical connection but disconnect/connect to the TCP server is done on DroneCore's end. So DroneCore could retry connecting to the same IP and port again, my TCP server keeps listening.

It would also be just fine if DroneCore removes dead connections completely from its connections list and `add_tcp_connection()` has to be called again.

julianoes, 2017-12-18:
Ok, I guess DroneCore should therefore internally try to reconnect and avoid looping on `recv()`. I'll try to fix that.

julianoes, 2017-12-18:
@fklabach I don't know how to test the TCP connection myself, so it would be great if you could give #203 a test.

Or if you could give me instructions how I can reproduce it that would also be helpful.

fklabach, 2017-12-18:
I've just tried #203 and it works well with my test code. DroneCore tries to reconnect at 1sec interval.

This is how you I can reproduce the issue:

- start a TCP server

- add a connection to DroneCore using `add_tcp_connection()` with IP/port of TCP server

- on TCP server, accept the TCP connection from DroneCore

- close the previously accepted TCP connection. This is where you get 100% CPU load.

julianoes, 2017-12-18:
Could you share your TCP server test code? That could be helpful for future debugging. As far as I know PX4 Firmware does not support TCP, so I'm not sure how this could be tested.

So, is this fix good enough for you?

hamishwillee, 2017-12-18:
+1 to the "Could you share your TCP server test code?". Would be great to have something we could add to CI for testing. 

@julianoes if not that, I wonder if anyone might have time and energy to write a drop in tcp replacement for the mavlink_main.cpp.

fklabach, 2017-12-18:
> So, is this fix good enough for you?

More than that, couldn't get a better fix for that issue.

I'm not sure if my code is of any help for your automated tests, it's written in Qt and really just starts a TCP server. You could still use it to check this issue yourself, I've attached the source and project file.
[dronecore_tcptest.zip](https://github.com/dronecore/DroneCore/files/1569623/dronecore_tcptest.zip)

julianoes, 2017-12-19:
Awesome, thanks for the feedback and test code @fklabach.

PR #203 (2017-12-19): core: try to reconnect TCP connection
This should avoid busy looping when a the TCP connection is dropped.

This could fix #201.
closed 2017-12-19 by julianoes