mavlink/MAVSDK #205: ArduPilot/other mavlink autopilot compatiblity/support

khancyr, 2017-12-19:
Hello, 

Do you plan to open compatibility/support for other mavlink based autopilot like ardupilot ?
Will you accept a PRs in that objectifs ?

Kiwa21, 2017-12-19:
That would be real nice IMO !

hamishwillee, 2017-12-19:
We would love to support other MAVLink autopilots, but:
1. Our main priority is rock solid performance on Dronecode platform. 
2. We don't have time or resources to maintain the port.

So the short answer is yes, but only if the person making the contributions commits to maintaining them and helps to put in place the infrastructure needed to test them. 

As an aside I think that as long as the `DroneCore` and `Device` classes work OK with another autopilot, the new architecture would give us lots of choices about how we might support said autopilot in the plugins.

@julianoes Is that a fair assessment of our position?