mavlink/MAVSDK #214: Device never gets connected again after heartbeat timed out

fklabach, 2018-01-04:
When heartbeats time out of a connected `Device`, the `Device` gets disconnected and never connected again. So both `is_connected()` (with only one device) and `is_connected(uuid)` (with uuid of previously connected device) never return true again. This also affects at least the `Device` thread update rate (100ms instead of 10ms).

Looking at the `device_impl` code, ` set_disconnected()` resets `_target_uuid_initialized`, but not `_target_uuid`. Now autopilot version is not requested again in `process_heartbeat()`, because `_target_uuid` was previously set to a nonzero value. Target uuid is never initialized and `Device` never gets connected again.

Fixes I tried, both work for me:

- `set_disconnected()` resets `_target_uuid_initialized`, `_target_uuid` and `_target_uuid_retries`.

-  `set_disconnected()` just doesn't reset  `_target_uuid_initialized`, keep the uuid of the prevously connected device (an obtained uuid never changes in my application, this might not work for others).

julianoes, 2018-01-04:
@fklabach thanks for the issue.

> `set_disconnected()` just doesn't reset _target_uuid_initialized, keep the uuid of the prevously connected device (an obtained uuid never changes in my application, this might not work for others).

Could you make a pull request for this suggestion? I'd say this makes more sense.

julianoes, 2018-01-04:
@fklabach When testing your fix I realized that I could not actually reproduce the problem that you see. Does your vehicle actually send a UUID? Or does it send 0? When I test it with PX4 SITL it seems to reconnect just fine.

fklabach, 2018-01-04:
I use PX4 SITL too (with Gazebo) and it sends a nonzero uuid.
Does your SITL send autopilot version at regular intervals? For me, autopilot version is only sent when requested.
Once `_target_uuid` was set nonzero, autopilot version is never requested again in `process_heartbest()`.
To get connected again would require `process_autopilot_version()` to set `_target_uuid_initialized`. This never happens when we  don't request and receive the autopilot version.

julianoes, 2018-01-04:
Ok I think I understand, thanks.

PR #215 (2018-01-04): Keep previous uuid when Device disconnects
This should fix #214.
closed 2018-01-04 by julianoes