mavlink/MAVSDK #216: Unable to run integration tests

shakthi-prashanth-m, 2018-01-04:
I am unable to run integration tests.
```
user@ubuntu:~/karma/fork/test/DroneCore$ ./build/default/dronecore-integrationtests --gtest_filter=FollowMeOneLocation
Note: Google Test filter = FollowMeOneLocation
[==========] Running 0 tests from 0 test cases.
[==========] 0 tests from 0 test cases ran. (0 ms total)
[  PASSED  ] 0 tests.
user@ubuntu:~/karma/fork/test/DroneCore$ ./build/default/dronecore-integrationtests --gtest_filter=TelemetryAsync
Note: Google Test filter = TelemetryAsync
[==========] Running 0 tests from 0 test cases.
[==========] 0 tests from 0 test cases ran. (1 ms total)
[  PASSED  ] 0 tests.
```
But it works if I use `*`.
```
user@ubuntu:~/karma/fork/test/DroneCore$ ./build/default/dronecore-integrationtests --gtest_filter=*
[==========] Running 30 tests from 2 test cases.
[----------] Global test environment set-up.
[----------] 23 tests from SitlTest
[ RUN      ] SitlTest.TwoConnections
Not autostarting SITL (use AUTOSTART_SITL=1)
```
Am I missing something ?

julianoes, 2018-01-04:
Check this: https://docs.dronecore.io/en/contributing/test.html#on-a-real-vehicle

You need:
```
./build/default/dronecore-integrationtests --gtest_filter="SitlTest.TelemetryAsync"
```

shakthi-prashanth-m, 2018-01-05:
Oh! OK.. I missed prepending `SitlTest.`. Thanks @julianoes .

closed 2018-01-05 by shakthi-prashanth-m