mavlink/MAVSDK #222: Rename Telemetry APIs that does status update

shakthi-prashanth-m, 2018-01-10:
These [Telemetry APIs](https://github.com/dronecore/DroneCore/blob/develop/plugins/telemetry/telemetry.h#L421-L608) are subscriptions to receive updates on the requested telemetry info at a specified interval. But when we no longer need such an update, as I know, we have to call again same API by passing `nullptr` for the callback parameter (for eg: as I did it [in case of Follow Me](https://github.com/dronecore/DroneCore/pull/217/files#diff-231fc8390befabbdecf5ac6bd62b0408R92)).
I think we can improve this by having a `unsubscribe_*()` or `de_register_*()` which are familiar among other platform developers. So, consequently we can change both subscribe & UN-subscribe APIs. 

Moreover, an API essentially be named after its functionality  So, I prefer to have them as below:
```
void request_flight_mode_updates(flight_mode_callback_t callback);
void remove_flight_mode_updates();
```
OR
```
void subscribe_flight_mode_updates(flight_mode_callback_t callback);
void unsubscribe_flight_mode_updates();

```

hamishwillee, 2018-01-10:
> s I know, we have to call again same API by passing nullptr for the callback parameter 

OK, I didn't know that and it isn't documented. I don't think it matters so much what APIs we use for turning off callbacks as long as we are consistent across the whole of the core library and document this as a "standard".

Personally I actually prefer our current naming over your suggestion:
```
device.telemetry().flight_mode_async(callback); //enable callbacks
device.telemetry().flight_mode_async(nullptr); //disable callbacks
```
OR
```
device.telemetry().flight_mode_async(callback); //enable callbacks
device.telemetry().cancel_flight_mode_async(); //disable callbacks
```
I just think they feel "less wordy", and I don't think people find them confusing provided that we are consistent. 

I would go with the majority.

julianoes, 2018-01-11:
I would keep what we have now or go with @shakthi-prashanth-m's suggestion. Both are fine with me, I think `subscribe_` and `unsubscribe_` are action clearer (and also what we currently use in the manually derived wrappers. However, I'm not sure if it is worth changing it all right now since there will be some churn due to gRPC anyway.

shakthi-prashanth-m, 2018-01-11:
Thanks @hamishwillee @julianoes  for sharing your opinions.
> I'm not sure if it is worth changing it all right now since there will be some churn due to gRPC anyway.

I agree that some things will change due to gRPC. There are around 15 of them that require the change.

julianoes, 2024-05-27:
We now use subscribe and unsubscribe methods.

closed 2024-05-27 by julianoes