mavlink/MAVSDK #224: DroneCore shell, a command line tool for interacting with a MAVLink device

shakthi-prashanth-m, 2018-01-13:
This is about proposal of **DroneCore shell**, a command line tool for interacting with a MAVLink device using DroneCore library. Its will be a handy tool for both users & developers.
May be this was discussed sometime back in the community. I am just recording it here for tracking later.

I feel that its very suitable, if someone wants to try DroneCore quickly like,
```
$ dronecore action { arm | disarm | takeoff | land} [args]
```
```
$ dronecore mission { upload | download | set | } [args]
```
etc.

I imagine that such tools are going to become familiar with many users in future, as we see such powerful tools like **curl**/**wget** in URL transfer domain. I think with similar thought _mavros_ command line (`mavcmd`) was introduced ( I think so).

After python bindings in place, we may think of developing such tools relatively easily.
I shared this with @julianoes earlier. I just wanted to hear what is the feedback in the community.

Please share your thoughts.
Thanks.

julianoes, 2018-01-15:
While this could be handy, I would vote to support easy scripting like this just in the Python library. The benefit that Python gives us is that you can inspect functionality using `help()` and it handles all command line parsing for you, etc.

JonasVautherin, 2018-05-30:
I believe that the python bindings will naturally enable easy scripting, and making a DroneCore shell package would be a separate (subproject) of DroneCore.

I'll close it for now, as I believe that will come back with the python bindings (but on the python repo, then :)).

julianoes, 2018-06-01:
I think that makes sense. We don't need to re-invent scripting.

closed 2018-06-01 by julianoes