mavlink/MAVSDK #226: Error during make default

Deanatius, 2018-01-16:
Hello, I have issues when trying to install DroneCore on my Jetson TX1. After doing the previous steps, I proceeded to make default and it start normally.

However, when it reaches _'Linking CXX shared library libdronecore.so'_ this error occurs.

/usr/bin/ld: cannot find -ltinyxml2
collect2: error: ld returned 1 exit status
CMakeFiles/dronecore.dir/build.make:1056: recipe for target 'libdronecore.so' failed
make[3]: *** [libdronecore.so] Error 1
CMakeFiles/Makefile2:179: recipe for target 'CMakeFiles/dronecore.dir/all' failed
make[2]: *** [CMakeFiles/dronecore.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make[1]: *** [all] Error 2
Makefile:71: recipe for target 'default' failed
make: *** [default] Error 2

I am not sure what causes the error as I have followed the steps carefully. I have checked the DroneCore/libs directory and the tinyxml2 folder does exist.

shakthi-prashanth-m, 2018-01-16:
Hi @Deanatius , did you build on [latest develop](https://github.com/dronecore/DroneCore) ?
>/usr/bin/ld: cannot find -ltinyxml2

This is mostly due to improper submodule update. Can you please double-check if you missed steps from [building library section](https://docs.dronecore.io/en/contributing/build.html#linux) as summarized below:

- Clone and update the submodules of DroneCore repository:
```
git clone https://github.com/dronecore/DroneCore.git
cd DroneCore
git submodule update --init --recursive
```
- Build the (debug) C++ library:
```
make default
```
I recommend to follow these steps and let us know.
Thanks.

Deanatius, 2018-01-16:
> Hi @Deanatius , did you build on latest develop ?
Yes I am building on the latest develop.

I am followings the steps and there are no error messages popping up during the submodule update.

 > However, when it reaches 'Linking CXX shared library libdronecore.so' this error occurs.
> /usr/bin/ld: cannot find -ltinyxml2
> collect2: error: ld returned 1 exit status
> CMakeFiles/dronecore.dir/build.make:1056: recipe for target 'libdronecore.so' failed
> make[3]: *** [libdronecore.so] [...]

I tried to build on my PC and everything works out well but the same error persists on my TX1.

julianoes, 2018-01-16:
@Deanatius for "native" installation (I guess the Jetson counts as native), you need to have libtinyxml2 installed.

```
sudo apt-get install libtinyxml2-dev
```

To see all dependencies, you can also check the [Dockerfile](https://github.com/dronecore/DroneCore/blob/develop/docker/Dockerfile-Ubuntu-16.04), or check the [docs](https://docs.dronecore.io/v/develop/en/contributing/build.html#linux) (it's only mentioned in the developer branch).

Deanatius, 2018-01-17:
> @Deanatius for "native" installation (I guess the Jetson counts as native), you need to have libtinyxml2 installed.
> 
> > sudo apt-get install libtinyxml2-dev

This did the trick! Sorry for my incompetence as I am a complete beginner in developing with Ubuntu and also with UAVs.

closed 2018-01-17 by Deanatius

julianoes, 2018-01-17:
@Deanatius no worries! Thanks for the feedback.