mavlink/MAVSDK #231: Unable to establish serial connection to PC

Deanatius, 2018-01-19:
Hello, 
I am currently developing using DroneCore and a Jetson TX1. I have managed to establish a serial connection via USB cable from the Pixhawk to the TX1 using:

` add_serial_connection("/dev/ttyACM0", 19200)`

However, when I connect to my PC using the same settings it failed to establish a connection. Am I doing something wrong or does it require different settings?

Yes, I have the ACM module installed on my PC.

julianoes, 2018-01-19:
When it fails, is there any error message?
Can you try to connect on the PC using screen or putty?
Are you sure that `/dev/ttyACM0` is correct for the PC?
Have you checked that ModemManager is not using the connection?

closed 2018-01-23 by Deanatius

julianoes, 2018-01-23:
@Deanatius could you comment on what resolved this? So that someone finding this in the future knows what it was? Thanks.

Deanatius, 2018-01-24:
I tried running the original program as root but still it did not work.

I changed the SYS_COMPANION settings in QGroundControl to Companion Link (921600 baud, 8N1) and also change baud rate in the code accordingly:
`add_serial_connection("/dev/ttyACM0", 921600)`
Ran the program as root again and it worked. Tried running the new program without _sudo_ and the same error occurred so could be those 2 factors.

julianoes, 2018-01-24:
@Deanatius thanks for the feedback.

To avoid having to use root, you need to add your user to the `dialout` group, at least on Ubuntu.

```
sudo usermod -a -G dialout $USERNAME
```