mavlink/MAVSDK #233: Flight testing DroneCore examples on Intel Aero

mrpollo, 2018-01-22:
@PX4/testflights  can you please help us verify the following documentation examples run on the Intel Aero:

* [Takeoff and Land](https://github.com/dronecore/DroneCore/tree/develop/example/takeoff_land)
* [Fly mission](https://github.com/dronecore/DroneCore/tree/develop/example/fly_mission) [Current example in master branch uses hard-coding. You can use [this patch](https://github.com/dronecore/DroneCore/tree/improve-fly-mission/example/fly_mission) which flies QGroundControl Mission plan]
* [Offboard velocity control](https://github.com/dronecore/DroneCore/tree/develop/example/offboard_velocity)
* [GPS Follow Me](https://github.com/dronecore/DroneCore/tree/develop/example/follow_me)

Please try to follow the steps in our docs to build and run the examples
* [Building DroneCore from source](https://docs.dronecore.io/en/contributing/build.html)
* [Running examples](https://docs.dronecore.io/en/examples/)

We are looking for all levels of feedback, from "steps are wrong, we can't build" to "vehicle flew in the wrong direction".

@shakthi-prashanth-m @anitha-suresh @julianoes @JonasVautherin how would you guys like the feedback from the test team? logs? command line output? screenshots?

mrpollo, 2018-01-22:
@PX4/testflights

shakthi-prashanth-m, 2018-01-23:
@mrpollo Thanks for creating the issue.
IMO, Basically these example should work as expected.
In either success/failure, it would be good to have:

- DroneCore logs (which appears on `stdout`)
- PX4 logs (`ulog`)
- Video of the examples run (possible ?)

shakthi-prashanth-m, 2018-01-23:
>•Fly mission [Current example in master branch uses hard-coding. You can use this patch which flies QGroundControl Mission plan]

@mrpollo I corrected **this patch** link which was referencing fly mission example in the master. It was supposed to refer https://github.com/dronecore/DroneCore/pull/232.

r0gelion, 2018-02-02:
I built the sample, but when I run the example:
`./takeoff_and_land`

I got the message that the device wasn't found, but I have a UDP connection to the vehicle since I'm seeing the telemetry data in QGC on the same computer running the example.

Any thoughts?

shakthi-prashanth-m, 2018-02-02:
@r0gelion Can you please add below UDP Endpoint in `/etc/mavlink-router/files/main.conf` ?

```
[UdpEndpoint dronecore]
Mode = Normal
Port = 14540
```
And then, execute
```
systemctl restart mavlink-router
```
Now, run `takeoff_and_land` example.

Thanks

r0gelion, 2018-02-03:
@shakthi-prashanth-m 

As soon as I restarted the mavlink-router with the UDP Endpoint added to main.conf, I lost telemetry data in QGC and I got the same previous message when I try to run the takeoff_and_land example.

[image]

Regards.

avinash-palleti, 2018-02-05:
@r0gelion 
Address field is missing in the above entry which you added in mavlink-router conf file, so by default it will take local host address. I guess you are running dronecore on a different system which is connected to Aero over wifi hotspot.
Can you please add below entries in main.conf,

```
[UdpEndpoint dronecore_wifi]
Mode = Normal
Port = 14540
Address = 192.168.8.255

[UdpEndpoint dronecore_usb]
Mode = Normal
Port = 14540
Address = 192.168.7.55

```

Please run below commands after updating main.conf,
`systemctl daemon-reload`
`systemctl restart mavlink-router`

r0gelion, 2018-02-06:
@avinash-palleti 

I got telemetry data again after the changes, executed the example but got some errors, unfortunately, I don't have a screenshot, I decided to update the repository and tried again.

Now I'm getting this error when executed the example
[image]

julianoes, 2018-02-07:
Oh, uh, that's because of the recent refactor.

It's addressed for Android in #252 but not for the general case here yet.

shakthi-prashanth-m, 2018-02-10:
@julianoes will your latest tinyxml2 fix on Mac resolve @r0gelion issue ?
If so we can request @r0gelion to retry.
@r0gelion please use #235  for testing fly mission. I updated the link in the example list above.
You can run using `./fly_qgc_mission` which flies default QGC mission plan.

r0gelion, 2018-02-13:
for take off and land example I still get the same error.

I'll check the #235

julianoes, 2018-02-13:
@JonasVautherin I guess for Mac desktop for now we could just fix this by creating a static library again.

JonasVautherin, 2018-02-13:
Yeah let's do that.

julianoes, 2018-02-15:
@r0gelion please try again, it might work now.

r0gelion, 2018-02-15:
@julianoes 

Still getting the same error, these are the steps I followed:

- Deleted the repository
- Cloned the repository
- Updated the modules (git submodule...)
- install the library (make default install)
- compiled the take off example
- ran the example

Still getting the libtinyxml2d.5.dylib not loaded error.

julianoes, 2018-02-15:
Thanks, that's weird though but I'll try to reproduce and let you know.

julianoes, 2018-02-15:
Thanks, that's weird though but I'll try to reproduce and let you know.

Can you try to do `sudo rm /usr/local/lib/libdronecore.dylib`? That might be leftover.

julianoes, 2018-02-15:
No, I'm wrong, I can reproduce it. Will fix it tomorrow, sorry about that.

closed 2018-02-19 by JonasVautherin

shakthi-prashanth-m, 2018-02-20:
Thanks for the fix @julianoes and @JonasVautherin.
@r0gelion can you please test on latest and confirm ? Thanks.

r0gelion, 2018-02-20:
@shakthi-prashanth-m @julianoes @JonasVautherin 

No, still same error, let me know what information do you need to help you to debug this issue.

Regards.

julianoes, 2018-02-20:
@r0gelion that's with a clean build and fresh install?
Also make sure to clean out anything in `/usr/local/lib/libdrone*`.

r0gelion, 2018-02-20:
@julianoes 

I deleted the Dronecore folder
Cloned the repository
Updated the submodules
Installed the library (make default install)
Compiled the example and executed it

julianoes, 2018-02-20:
Try:
`sudo rm /usr/local/lib/libdronecore*` to clean up installed libs.

r0gelion, 2018-02-20:
That worked, I'm not getting the error anymore. I'll test the example and let you know.

Regards.

julianoes, 2018-02-20:
@r0gelion cool thanks for checking

r0gelion, 2018-02-22:
I got some errors when running the example, check for them here: http://pasted.co/78a4dfae

I'm also seeing the connection to QGC is taking very long time and it ends up in errors to get the parameters:
[image]

Let me know if you have an idea of what could be happening.

Thanks!

shakthi-prashanth-m, 2018-02-23:
@r0gelion which version of aero do you have ? Kindly paste output of `aero-get-version.py`.
Thanks.

julianoes, 2018-02-23:
> I'm also seeing the connection to QGC is taking very long time and it ends up in errors to get the parameters

I think this means that something about the param caching fails in QGC. Does it happen everytime? Is this with latest master and a recent QGC?

r0gelion, 2018-02-23:
It happens every time, it was yesterday's master and yesterday's daily build of QGC.

PR #263 (2018-02-15): build_system: create a static lib for Mac for now
This should resolve linking issues with examples. Once we have it all
packaged, we can change it again.

Fixes #233.
PR #276 (2018-02-19): build_system: build statically again for Mac
This is just an interim fix in the meantime.

Should fix #233.