mavlink/MAVSDK #234: Arming Rejected

Deanatius, 2018-01-23:
I have checked that all the pre-arm checks are passed and I am able to arm through controller and QGroundControl.  However, when I try to arm through DroneCore I get the error "Arming failed: Command denied"

`Telemetry::Health check_health = device.telemetry().health();
    bool calibration_required = false;
    if (!check_health.gyrometer_calibration_ok) {
        cout << ERROR_CONSOLE_TEXT << "Gyro requires calibration." << NORMAL_CONSOLE_TEXT << endl;
        calibration_required=true;
    }
    if (!check_health.accelerometer_calibration_ok) {
        cout << ERROR_CONSOLE_TEXT << "Accelerometer requires calibration." << NORMAL_CONSOLE_TEXT << endl;
        calibration_required=true;
    }
    if (!check_health.magnetometer_calibration_ok) {
        cout << ERROR_CONSOLE_TEXT << "Magnetometer (compass) requires calibration." << NORMAL_CONSOLE_TEXT << endl;
        calibration_required=true;
    }
    if (!check_health.level_calibration_ok) {
        cout << ERROR_CONSOLE_TEXT << "Level calibration required." << NORMAL_CONSOLE_TEXT << endl;
        calibration_required=true;
    }
    if (calibration_required) {
        return 1;
    }

    // Arm vehicle
    cout << "Arming..." << endl;
    const Action::Result arm_result = device.action().arm();

    if (arm_result != Action::Result::SUCCESS) {
        cout    << ERROR_CONSOLE_TEXT << "Arming failed:" << Action::result_str(arm_result) 
                << NORMAL_CONSOLE_TEXT << endl;
        return 1;
    }

    sleep_for(seconds(3));

    // Disarm vehicle
    cout << "Disarming..." << endl;
    const Action::Result disarm_result = device.action().disarm();`

This is abstract of the code I wrote just to get asynchronous updates for the flight mode and altitude for 3 seconds. The health checks passed as usual (did not perform GPS checks because flying indoors).
What are the other causes of arming fail? Do I need to manually set the flight mode in the program before arming?

hamishwillee, 2018-01-29:
Hi @Deanatius, 

As a matter of interest, are you using the master or develop branch?

DroneCore does not check the USB. 

What do you get out if you run the whole health check as per here: https://docs.dronecore.io/v/develop/en/guide/taking_off_landing.html#health-check

Can you also print out the [mode](https://docs.dronecore.io/v/develop/en/api_reference/classdronecore_1_1_telemetry.html#classdronecore_1_1_telemetry_1a881d44b3a1522ea14bff8834edd4145a)? 

It is possible you are in a mode that requires position infomration (GPS) and/or home position to be set and arming is therefore prevented. It may also be that you have set that RC is required and it is not enabled. This is probably a better question therefore to ask in discussion forums: http://discuss.px4.io/

julianoes, 2018-02-01:
> The health checks passed as usual (did not perform GPS checks because flying indoors).

DroneCore as of now does not support flying without GPS. It is only designed for "somehwat" autonomous modes and therefore it requires a position estimate.

Before arming, it tries to switch to HOLD mode which fails. I think the right thing to do for now, is to add a specific error messages saying "no position estimate".

hamishwillee, 2018-02-01:
@julianoes What would we need to do to support flying without GPS? Do we need a big fat warning on our readme/intro about this limitation?

Deanatius, 2018-02-02:
> Before arming, it tries to switch to HOLD mode which fails.

Figured as much. Currently, I am manually arming through the controller before running an Offboard program through the companion computer. However, when exiting OFFBOARD mode it switches to HOLD mode which is again not possible indoors. 

I read the documentation that it is possible to exit OFFBOARD mode by switching modes using the API. May I get an example code of how this is done?

julianoes, 2018-02-02:
@hamishwillee I vote that DroneCore never supports anything without position estimate. It doesn't need to be a global (latitude/longitude) position but at least a local position estimate. DroneCore is for the future and not for manual systems that become legacy in my opinion.

hamishwillee, 2018-02-05:
> @hamishwillee I vote that DroneCore never supports anything without position estimate. It doesn't need to be a global (latitude/longitude) position but at least a local position estimate. DroneCore is for the future and not for manual systems that become legacy in my opinion.

@julianoes While that is probably a good idea, the problem for me is that nowhere is it documented (AFAIK) what is required in terms of hardware/software to get a position estimate? So it is fine to say we don't support it, but we should point to docs that explain how support the main *use cases* (e.g. indoor flight). Thoughts?

Further to that, we should add this statement as an addition to the FAQ.

julianoes, 2018-02-05:
@hamishwillee that is true. We should document that for now it is GPS only and in the future it will also support flow or VIO.

hamishwillee, 2018-02-06:
@Deanatius So in summary - using DroneCore indoors is currently not supported (or more specifically, DroneCore without a position estimate) - https://github.com/dronecore/docs/pull/89. This is therefore not a bug but expected behaviour.

closed 2018-02-06 by hamishwillee