mavlink/MAVSDK #240: Internal team process

mrpollo, 2018-02-01:
Who are the team members?
How often do we meet?
How do we set direction for the project?

JonasVautherin, 2018-02-12:
Should we address that tomorrow during the call? =)

JonasVautherin, 2018-05-30:
What do we do with that? I believe we should either do something, or close it :-).

hamishwillee, 2018-05-31:
I think it is worth the discussion because I don't know (all) the answers. Am I a member of the team? Are you having meetings to discuss all this stuff that I wasn't invited to. [crying like a baby].

julianoes, 2018-06-01:
> Am I a member of the team?

Yes, definitely.

> Are you having meetings to discuss all this stuff that I wasn't invited to. [crying like a baby].

Yes, but the time we settled at which is latish for Europe and early for the West coast is not exactly great for Australia, so we assumed it wouldn't be worth it for you.

julianoes, 2020-11-20:
This is old, closing. For anyone wanting to get in touch with the MAVSDK maintainer team, just create an issue and the maintainers will respond.

closed 2020-11-20 by julianoes