mavlink/MAVSDK #248: Mission integration test SitlTest.MissionChangeSpeed is failing.

shakthi-prashanth-m, 2018-02-06:
Mission integration test [SitlTest.MissionChangeSpeed](https://github.com/dronecore/DroneCore/blob/develop/integration_tests/mission_change_speed.cpp#L33) is failing with below error.

First:
```
[10:32:52|Warn ] speed check, should be: 3 m/s, actually: 1.53522 m/s (mission_change_speed.cpp:87)
/home/intel/karma/fork/test/DroneCore/integration_tests/mission_change_speed.cpp:89: Failure
Expected: (speed_actual) > (speed_correct - 1.0f), actual: 1.53522 vs 2
```

Second:
```
[10:33:23|Warn ] speed check, should be: 8 m/s, actually: 5.62354 m/s (mission_change_speed.cpp:87)
/home/intel/karma/fork/test/DroneCore/integration_tests/mission_change_speed.cpp:89: Failure
Expected: (speed_actual) > (speed_correct - 1.0f), actual: 5.62354 vs 7
```
Can this be ignored or does it need a fix ?

julianoes, 2018-02-06:
> Can this be ignored or does it need a fix ?

Eventually this needs to be fixed, thanks for tracking this.

julianoes, 2019-04-30:
Addressed in #642.

JonasVautherin, 2019-04-30:
I think github takes it the other way round: if in #642 you write "Resolves #248", then merging #642 will close this one automatically :+1:.

julianoes, 2019-05-01:
True but I just wanted to note it anyway so that no one tries to solve it in the meantime.

julianoes, 2019-05-16:
Fixed or rather ignored for now by #642.

closed 2019-05-16 by julianoes