mavlink/MAVSDK #250: Create new Mission

dangviethieu-hntv, 2018-02-06:
Hello all. I have a question. I create and fly mission successfully. Now I want to create new mission after finish old mission or land, How? I try callback upload_mission_async() function, but it error.
And how do device stop mission? It only have start_mission_async() and pause_mission_async() function.
Thanks very much.

julianoes, 2018-02-06:
> @dangviethieu-hntv:
> I try callback upload_mission_async() function, but it error.

Can you please paste the output of this error? And does this happen consistently?

julianoes, 2018-02-12:
@dangviethieu-hntv can you please follow up? Thanks.

julianoes, 2018-02-12:
@shakthi-prashanth-m could you at some point test if you can send and execute two missions after each other?

shakthi-prashanth-m, 2018-02-14:
@julianoes Sure I'll.

julianoes, 2018-03-05:
@shakthi-prashanth-m what's the result here? I guess this has been fixed?

shakthi-prashanth-m, 2018-03-06:
@julianoes Unfortunately, we don't exactly know what error was prompted. I created branch https://github.com/dronecore/DroneCore/compare/verify-multi-mission for testing this.

@dangviethieu-hntv Could you please provide console logs of the example you run ?
>And how do device stop mission?

I think even in QGC, you can only change Mission mode to some other mode for stopping a mission.
So, you may call `action->land()` that stops executing current mission item and does mode switch to `AUTO.LAND`.
As I understand your query, you want to fly different missions one after the other. Could you please try testing https://github.com/dronecore/DroneCore/commit/5aec81f36eef551f42c4d64b5c24598eb389dae3 ?

julianoes, 2018-03-06:
@shakthi-prashanth-m I think it would be great to have the multi mission test as an integration test.

shakthi-prashanth-m, 2018-03-06:
>@shakthi-prashanth-m I think it would be great to have the multi mission test as an integration test.

OK

julianoes, 2018-07-24:
Work in progress: #442

JonasVautherin, 2019-06-11:
Closing for inactivity, not sure if #442 solved it though.

closed 2019-06-11 by JonasVautherin

julianoes, 2019-06-11:
Yes fixed in #642.