mavlink/MAVSDK #254: Build in docker as user

julianoes, 2018-02-08:
Currently when we build in docker it uses root and therefore leaves build artefacts with root as the owner. This requires `sudo make distclean` to clean up afterwards outside of docker which is inconvenient.

As @hamishwillee suggests we should build something similar to:
https://dev.px4.io/en/test_and_ci/docker.html
https://github.com/PX4/containers/blob/master/docker/px4-dev/Dockerfile_base#L64

dagar, 2018-02-08:
Docker ENTRYPOINT with this script seems to work well.
https://github.com/PX4/containers/blob/master/docker/px4-dev/scripts/entrypoint.sh

Dockerfile

    # create and start as LOCAL_USER_ID
    COPY scripts/entrypoint.sh /usr/local/bin/entrypoint.sh
    ENTRYPOINT ["/usr/local/bin/entrypoint.sh"]

When running the container you need to pass - eLOCAL_USER_ID=$UID and it creates an internal user matching your UID. Otherwise it uses UID 9001, which is still better than root.

julianoes, 2018-02-08:
@dagar last time I tried to use that it did not work for me, I remember. I'll have to try it again.

commit 79bf5dac (2018-09-21): Fix #254 Run docker build enviroment as normal user. Add convenience script in root directory to invoke docker container. Add README in docker folder describing how to build the image and run the containers.
closed 2018-09-21 by julianoes