mavlink/MAVSDK #272: Missing sample plan for example

hamishwillee, 2018-02-16:
https://github.com/dronecore/DroneCore/tree/develop/example/fly_qgc_mission

Does not contain "qgroundcontrol_sample.plan" as described in example: "(b) Use a pre-created sample mission plan in "example/fly_qgc_mission/qgroundcontrol_sample.plan"."

It would be good if this could be added :-)

hamishwillee, 2018-02-16:
Actually, I see you have a plan here: https://github.com/dronecore/DroneCore/blob/develop/plugins/mission/qgroundcontrol_sample.plan

So perhaps just update the documentation in the example?

shakthi-prashanth-m, 2018-02-16:
Yes! It was moved later. I'll update the example. Thanks!

shakthi-prashanth-m, 2018-02-16:
@hamishwillee, Do you think this link is useful ? If not, I shall remove it.

hamishwillee, 2018-02-19:
> @hamishwillee, Do you think this link is useful ? If not, I shall remove it.

Yes, I think removing it is a good idea.

PR #273 (2018-02-19): Update file path of QGC sample plan in the example
Update path of sample plan: `plugins/mission/qgroundcontrol_sample.plan`.
Fixes #272
closed 2018-02-19 by hamishwillee