mavlink/MAVSDK #279: Why don't we use exception handling for error handling  ?

shakthi-prashanth-m, 2018-02-20:
As per my understanding, exceptional handling makes code readable and maintainable;
thus our examples or applications can be even simpler.

// Takeoff land example may look like this:
[image]

hamishwillee, 2018-02-20:
The example wouldn't be quite this simple - you still have to wait for arm to complete before doing takeoff, and takeoff to complete before doing land. I'd also much rather wait on arming than have an exception.

Would exceptions play nice with the protobuf stuff?

shakthi-prashanth-m, 2018-02-20:
>you still have to wait for arm to complete before doing takeoff, 

We don't have to wait for arm to complete in sync version; we may have to, in async.
>and takeoff to complete before doing land

Yes, we do need to wait for the vehicle to reach takeoff altitude; I think its reasonable to let users to pass an optional argument `max_wait_time_s` to do the waiting inside API (for being sync version); we do so [in setting Follow Me configuration](https://github.com/dronecore/DroneCore/blob/48d74df8f3e04f773c6007b95418129ab80e9a09/plugins/follow_me/follow_me_impl.cpp#L96). 

@hamishwillee @julianoes @JonasVautherin what is your opinion ?

gRPC seems to support it:
- [Java](https://grpc.io/docs/quickstart/java.html#update-the-client) 
- [Android Java](https://grpc.io/docs/quickstart/android.html#update-the-client) 
- mostly [C++](https://github.com/grpc/grpc/blob/master/test/cpp/end2end/exception_test.cc)

@JonasVautherin do you know ?

julianoes, 2018-02-20:
@shakthi-prashanth-m there are pros and cons to using exceptions in C++.
https://hackernoon.com/error-handling-in-c-or-why-you-should-use-eithers-in-favor-of-exceptions-and-error-codes-f0640912eb45
https://stackoverflow.com/questions/253314/exceptions-or-error-codes
https://google.github.io/styleguide/cppguide.html#Exceptions

I decided to go without exceptions when I started it. Initially the C++ API needed to be simple and clean in order that it can easily be wrapped (manually) by other languages.

Right now I'm very reluctant to change everything that we have so far. I think we should stabilize the gRPC work and refrain from other refactoring that aren't absolutely necessary.

shakthi-prashanth-m, 2018-02-20:
@julianoes OK, I get it. Thanks for the clarification.

JonasVautherin, 2018-02-20:
I agree with @julianoes :-).

julianoes, 2018-02-22:
Ok, I'll close this then, thanks @shakthi-prashanth-m for raising this though, appreciated.

closed 2018-02-22 by julianoes