mavlink/MAVSDK #284: Example for installing the DroneCore on Raspberrypi

XiaoHuiGitHub, 2018-02-25:
It's necessary to present an example for installing the DroneCore on the RaspberryPi. 
I have tried to build the DroneCore on the Rpi step by step according to “Building DroneCore from Source on Linux” in the Docs. But there is something wrong when implementing "make default". The "/usr/bin/ld: cannot find -ltinyxml2"  is reported  when Linking CXX shared library libdronecore.so.
Does it  mean I have to install tinyxml2 on the raspberrypi firstly? 
[image]

hamishwillee, 2018-02-25:
Hi @XiaoHuiGitHub 

The short answer is yes. 

I think you are installing the _develop_ branch using the _master_ branch instructions. Please follow the _develop_ branch instructions here: https://docs.dronecore.io/v/develop/en/contributing/build.html
These include libxml instructions.

FYI We have source 2 branches - "master" and "develop". The master branch is supposed to reflect the current stable version of DroneCore while the develop is the head revision. Unfortunately the default source tree on Github that you get is "develop" while the default documents are for "master". This is a bit of a problem now because the master and develop revisions have diverged quite a lot.

closed 2018-02-25 by hamishwillee

XiaoHuiGitHub, 2018-02-26:
Dear @hamishwillee,
I have solved such problem about "/usr/bin/ld: cannot find -ltinyxml2" by installing the tinyxml2 lib on the RaspberryPi individually. And I have tried change the 2 branches, it's not the reason. 

We have to install the tinyxml2 using **_sudo apt-get install libtinyxml2-dev_**. 

The results are shown here.
[image]

hamishwillee, 2018-02-26:
@XiaoHuiGitHub Perhaps I wasn't clear. You need to use the `develop` branch, and follow the instructions in the [develop version of the guide](https://docs.dronecore.io/v/develop/en/contributing/build.html#linux).

The very first step there is this line:
```
sudo apt-get install cmake build-essential colordiff astyle git libcurl4-openssl-dev libtinyxml2-dev doxygen -y
```

As you can see, this includes installing `libtinyxml2-dev`.

XiaoHuiGitHub, 2018-02-26:
Hey @hamishwillee ,

Thank you very much, I got it. 
It's really strange that I have implemented the very first step just as you showed above.

Than I want to use the  RaspberryPi with DroneCore to control a real vehicle. 
Would you please give me some suggestions or references ?

julianoes, 2018-02-26:
> Than I want to use the RaspberryPi with DroneCore to control a real vehicle.
Would you please give me some suggestions or references ?

For that please create a new issue for a specific question, or ask on the PX4 slack.

XiaoHuiGitHub, 2018-02-27:
@julianoes，thank you! 
It's a good ideal and I will create a specific topic about the experiments on a real vehicle after the communication tests.