mavlink/MAVSDK #285: [Issue] Yaw of mission item always is 0

dangviethieu-hntv, 2018-02-26:
Hi all. I have mission and I want drone fly with YAW is set by mission item. I config in QGroundControl as: parameter -> mission -> MIS_YAWMODE is "**Heading as set by waypoint**", and I set code misison item as: _new_item->set_gimbal_pitch_and_yaw(0.0f, 60.0f);_ . But when flying, YAW always is 0.
I reviewed code in: **DroneCode/plugins/mission/mission_item_impl.cpp** and saw **issue** in function:
`float MissionItemImpl::get_mavlink_param4() const {`
`float yaw_angle_deg = 0.0f;`
 `return yaw_angle_deg;`
`}`
I think **yaw_angle_deg** parameter should be set **yaw_angle_deg = _gimbal_yaw_deg**.
So, anyone please help me check it again and update if it necessary.
Many thanks so much !!!

hamishwillee, 2018-02-26:
@dangviethieu-hntv Feel free to create a PR for the change - we can then more easily discuss it. 

Otherwise @shakthi-prashanth-m or one of the other guys will hopefully be able to look at it soon (looks like you're probably right to me)

julianoes, 2018-02-26:
@dangviethieu-hntv we use the gimbal to set the yaw angle of the camera but not the drone itself. This means that by default we just want the drone to face forward and I believe that's achieved by setting `param4` to `0`.

This has the advantage that you fly forwards and you can use front facing obstacle avoidance if you have it.

dangviethieu-hntv, 2018-02-26:
@julianoes I want set the yaw angle of the drone itself. How?
If I fix get_mavlink_param4() function as: **yaw_angle_deg = _gimbal_yaw_deg**, Drone will fly with YAW set in _new_item->set_gimbal_pitch_and_yaw(float pitch_deg, float yaw_deg)_ function.
I understand when set MIS_YAWMODE in QGroundControl is: **Heading as set by waypoint** , and Drone will fly with YAW set in missionitem. So, Do we haven't **set_pitch_and_yaw** function?

julianoes, 2018-02-26:
@dangviethieu-hntv that's currently not supported I'm afraid.

What exactly is your use case? Why would you need it?

dangviethieu-hntv, 2018-02-26:
@julianoes example, I want drone fly follow and toward the cycling. So, I really need it.

julianoes, 2018-02-26:
@dangviethieu-hntv I don't understand, please explain this with a few more sentences, a picture, a video anything.

dangviethieu-hntv, 2018-02-26:
[image]
Thanks you very much with helpful. I want drone fly and always toward the cycling.

julianoes, 2018-02-26:
What's the cycling? I think this kind of flight behavior is better done using for instance offboard control where you plan a trajectory instead of waypoints.

dangviethieu-hntv, 2018-02-26:
It same the Cycle racing track. Drone will fly parallel and toward it. If fix in get_mavlink_param4() function yaw_angle_deg parameter. I will study about offboard and try it again.
Thanks you very much.

julianoes, 2018-02-26:
@dangviethieu-hntv I can see why you would need it but really the proper solution is a gimbal or offboard control for this.

DroneCore tries to enable the main use cases with a simple way out of the box in an opinionated way. However, the goal is not to allow each and every mavlink option because then it will (like mavlink) be very hard to figure out what is actually possible and supported. Therefore only a small set of features is included in the API, and that's why I'm pushing back on your request.

dangviethieu-hntv, 2018-02-26:
Oh, I will study gimbal or offboard control after.
Thanks you very much with helpful.

julianoes, 2018-02-26:
No worries!

closed 2018-02-26 by julianoes