mavlink/MAVSDK #301: Run Example Takeoff Fail

dangviethieu-hntv, 2018-03-04:
I ran take_off example and have issue: "Vehicle not ready to arm"
But I ran fly_mission example successfully.
Anyone can help me !!!

hamishwillee, 2018-03-05:
Best place for support queries is Slack - see https://docs.dronecore.io/en/#getting-help

Both examples take off in the same way, and will exit if not ready to arm. Probably something changed between the examples - e.g. you lost GPS. You can find out what is blocking arming by running a health check. There is some information in this in https://docs.dronecore.io/v/develop/en/guide/taking_off_landing.html#taking-off

closed 2018-03-05 by hamishwillee

shakthi-prashanth-m, 2018-03-05:
@dangviethieu-hntv , as suggested by @hamishwillee , replace `if` by `while`.
Vehicle may not be ready when checked in the first time.
```cpp
// Wait until health is OK and vehicle is ready to arm
while (telemetry->health_all_ok() != true) {
    std::cout << "Vehicle not ready to arm ..." << std::endl;
    std::this_thread::sleep_for(std::chrono::seconds(1));
}
```