mavlink/MAVSDK #321: Rendering of reference and pointer parameters in markdown script

hamishwillee, 2018-03-15:
Currently a parameter with a ref or pointer renders with a space after the ref like this: `Device & somedevice` or `Device * somedevice` (rendering is fine value parameters).

The question is, should this render as `Device& somedevice` or `Device &somedevice`

(as an aside, the second format is slightly harder, because for pointers, care needs to be taken to make sure the * is not interpreted as "make text italic".

This originates here: https://github.com/dronecore/docs/pull/107#discussion_r174682278

hamishwillee, 2018-03-15:
Which is best is a matter of style - have we standardized on this in source code?

shakthi-prashanth-m, 2018-03-15:
So far we've used this format in source code: `Device &somedevice`.

```cpp
plugins/follow_me/follow_me.h:    const Config &get_config() const;
plugins/follow_me/follow_me.h:    const TargetLocation &get_last_location() const;
plugins/follow_me/follow_me_impl.h:    const FollowMe::Config &get_config() const;
plugins/follow_me/follow_me_impl.h:    const FollowMe::TargetLocation &get_last_location() const;core/device.h:    Time &get_time() { return _time; };
core/device.h:    DroneCoreImpl &_parent;
core/dronecore_impl.h:    Device &get_device();
core/dronecore_impl.h:    Device &get_device(uint64_t uuid);
core/dronecore_impl.h:    const std::vector<uint64_t> &get_device_uuids() const;
core/dronecore.h:    const std::vector<uint64_t> &device_uuids() const;
core/mavlink_receiver.h:    mavlink_message_t &get_last_message()
core/mavlink_receiver.h:    mavlink_status_t &get_status()
...
```
I think, even `Device& somedevice` also fine: It says `somedevice` is of type `Device&`.
But `Device & somedevice` is hard to read. Isn't it ?

hamishwillee, 2018-03-15:
> But `Device & somedevice` is hard to read. Isn't it ?

IMO it is easy to read, but it looks a little unnatural. I have fixed this in https://github.com/dronecore/DroneCore/pull/324

It uses `Device& somedevice` for parameters. This is easier to implement and looks better in more cases.

FYI, the prototypes etc are still rendered with odd spaces. It would be a lot more effort to tidy these cases up because the whole strings are provided by doxygen.

hamishwillee, 2018-05-31:
Device &somedevice = Device &somedevice :-)

julianoes, 2018-06-01:
@hamishwillee thanks fixed above.

AuterionWrikeBot, 2019-06-13:
➤ Hamish Willee commented:

Julian Oes Long time since this discussion... and I thought it was "done". Can you remind me what action, if any, I am supposed to take here?

julianoes, 2019-06-13:
We should change it to `Device& somedevice` according to [CppCoreGuidelines](https://github.com/isocpp/CppCoreGuidelines/blob/master/CppCoreGuidelines.md#nl18-use-c-style-declarator-layout).

julianoes, 2019-06-13:
I can do the change after the rename.

hamishwillee, 2019-06-17:
Cool. So when the code is all "right" I'll have another look and see if I can match it in markdown. Note though, that this sort of thing is fiddly, and as long as the code is readable in markdown I do not see this as my highest priority.

PR #790 (2019-07-16): Change clang-format style
**AlignAfterOpenBracket -> AlwaysBreak**

The main reason for this change is to work around a clang-format bug in AlignAfterOpenBracked introduced with clang-format 8:
https://bugs.llvm.org/show_bug.cgi?id=41676

By changing to this option, we resolve the problem were different versions enforce different styles.

Also, in my opinion this style makes more sense in general.
For some argumentation by Kevlin Henney, I suggest to watch:
https://youtu.be/ZsHMHukIlJY?t=1027

**PointerAlignment -> Left**

We wanted to switch the pointer and reference alignment style for a whi [... cut]
closed 2019-07-16 by JonasVautherin